rtic-rs/rtic #178: [RFC] default all task capacities to 1

japaric, 2019-04-21:
## Summary

If the user didn't specify the capacity of a task default to a capacity of `1`.

## Motivation

Today when the capacity of a task is not specified the framework tries to pick
a capacity that reduces the chances of `spawn` / `schedule` failing at runtime.
To do so the framework assumes that (a) each context that may spawn / schedule
a particular task `X` will do so at most *once* per invocation and (b) that such
context runs not so often that the task will be dispatched before the context
runs again.

This works reasonably well sometimes and sometimes it can lead to more memory
usage than necessary. More importantly, though, it makes it hard to estimate the
capacity of a task from just its attribute. Quick, what are the capacities of
tasks `foo` and `bar` in the following program?

``` rust
#[app(device = /* .. */)]
const APP: () = {
    #[task(spawn = [bar])]
    fn foo(c: foo::Context) {
        // ..
    }

    #[task(spawn = [bar])]
    fn bar(c: bar::Context) {
        // ..
    }

    #[init(spawn = [bar])]
    fn init(c: init::Context) {
        // ..
    }

    extern "C" {
        fn EXTI1();
    }
};
```

The answer is `0` (yes, *zero*) and `3`, respectively. `foo` can never be
`spawn`-ed so its capacity is `0`; `bar` appears in three `spawn` lists so
its capacity is `3`. Modifying the program can make these numbers fluctuate,
making things even harder to predict.

Therefore I propose that the capacity of a task defaults to the smallest
(sensible) value of `1` when not specified. This behavior makes capacities fully
predictable.

## Design

Drop the logic used to guess an appropriate task capacity and simply default to
a value of `1` if the capacity is not specified by the user.

## Alternatives

We could ask the user to specify the capacity of all tasks, that is we could
have no default. However, this would burden the end user with deciding on a
capacity as soon as they add a new task. This could break the development flow
and hurt productivity; choosing the best / right capacity requires looking at
the program as a whole and it's likely best done in later stages of development.

korken89, 2019-04-28:
I think this is good, it's a sane default. I will give it my approval.
Forcing the user to put a value might be a bit over zealous IMO.

japaric, 2019-06-29:
This RFC seems uncontroversial. Let's give it a 1 week-ish FCP with disposition to merge it. 

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi

TeXitoi, 2019-06-29:
Ok for me

japaric, 2019-07-08:
:tada:  This RFC has been formally approved. Implementation is in PR #205 (I'm going to keep this open until the PR lands)

commit 2fa07c8c (2019-09-15): Merge #205

205: rtfm-syntax refactor + heterogeneous multi-core support r=japaric a=japaric

this PR implements RFCs #178, #198, #199, #200, #201, #203 (only the refactor
part), #204, #207, #211 and #212.

most cfail tests have been removed because the test suite of `rtfm-syntax`
already tests what was being tested here. The `rtfm-syntax` crate also has tests
for the analysis pass which we didn't have here -- that test suite contains a
regression test for #183.

the remaining cfail tests have been upgraded into UI test so we can more
thoroughly check / test the error message presented to the  [... cut]
PR #205 (2019-09-15): rtfm-syntax refactor + heterogeneous multi-core support
this PR implements RFCs #178, #198, #199, #200, #201, #203 (only the refactor
part), #204, #207, #211 and #212.

most cfail tests have been removed because the test suite of `rtfm-syntax`
already tests what was being tested here. The `rtfm-syntax` crate also has tests
for the analysis pass which we didn't have here -- that test suite contains a
regression test for #183.

the remaining cfail tests have been upgraded into UI test so we can more
thoroughly check / test the error message presented to the end user.

the cpass tests have been c [... cut]
closed 2019-09-15 by japaric