rtic-rs/rtic #184: timer-queue: need to unlock DWT

japaric, 2019-05-01:
> RTFM schedule doesn't work on STM32F7xx. After some test at Oxydize with Jorge, we found that the DWT timer is never started. It needs some more configuration.
By adding the line unsafe { (*DWT::ptr()).lar.write(0xC5ACCE55) }; it works.
It should be added to the cortex-m DWT init for STM32F7xx

Originally reported by @musitdev in rust-embedded/cortex-m#144

japaric, 2019-05-01:
@musitdev can you try PR #185 or #186 and see if that fixes this issue for you?

musitdev, 2019-05-02:
Ok I'll test.
To answer to @korken89 has I seen on the [internet](https://stackoverflow.com/questions/36378280/stm32-how-to-enable-dwt-cycle-counter), it's a Cortex M7 feature. On Cortex M7 the debug reg must be unlock with DWT->LAR = 0xC5ACCE55

japaric, 2019-05-02:
@musitdev please don't forget to remove the LAR.write you had in your init function when testing this.

@korken89 the DWT has an optional "software lock" feature (the ITM has something similar). I checked and it seems that the STM32F103 and the nRF52840 don't have this software lock feature; it seems that the STM32F7 does implement this feature.

@musitdev could you also check and report what's the value of DWT_LSR at reset? It's bit 0 should be 1, that indicates that the software lock feature is implemented.

korken89, 2019-12-29:
With v5 this should not be an issue anymore

closed 2019-12-29 by korken89