rtic-rs/rtic #195: `init` must have type signature `[unsafe] fn() [-> init::LateResources]`

HansHamster, 2019-05-25:
Hello,

I tried to get started by following the instructions [here](https://japaric.github.io/rtfm5/book/en/by-example/new.html) after setting up the target as described [here](https://docs.rs/cortex-m-quickstart/0.3.0/cortex_m_quickstart/), but (aside from the outdated link to the init.rs) I get stuck as soon as I try to build the project:
> error: `init` must have type signature `[unsafe] fn() [-> init::LateResources]`

I tried a few different examples as well as the current stable and nightly version of rust, but the problem remains.

Maybe I'm just doing something stupid on my end, but I have no idea what is wrong.

TeXitoi, 2019-05-25:
We can't help you if you don't give us the project you're trying to build and the commands you use to compile it.

HansHamster, 2019-05-25:
I just tried to build the example linked above.

But I found the problem... For whatever reason I was reading the instructions for an unreleased version of rtfm, but tried to build it using the current 0.4.3 release that cargo pulled...

No idea how I found those instructions and not the ones for the current release.

closed 2019-05-25 by HansHamster