rtic-rs/rtic #196: Schedule does not work properly in init after connecting power.

thalesfragoso, 2019-06-09:
This one is a bit weird. 
I set up a periodic task following the example from the book. The task is suppose to run once every 5ms. After connecting power, it looks like the task only starts running 10 seconds after init, and from there it runs properly (5ms period). But if I press the reset button just after connecting power everything works as it should.

I just encounter the problem if I completely disconnect power and reconnect and don't press the reset button. I changed the schedule in init to spawn and maintained the schedule inside the software task and now it works as it should even just after connecting power.

I really don't know why pressing reset would be that different from connecting power for the first time.

I am testing this on a stm32f103c8 (bluepill).

TeXitoi, 2019-06-09:
Please point to the complete code, else it's difficult to analyse.

TeXitoi, 2019-06-09:
Aldo what's your setup: do you power with a stlink? Plugged to z computer or a power supply?

thalesfragoso, 2019-06-09:
> Please point to the complete code, else it's difficult to analyse.

Sorry, I didn't post it because was almost the same as the example, but you are right, it is way easier to find a problem if you can look at the code. So, here it is most of it:

`const DEBOUNCE_PERIOD: u32 = 320_000;

#[app(device = stm32f1xx_hal::pac)]
const APP: () = {

    static mut PWM_HANDLER: Pwm<pac::TIM2, C1> = ();
    static mut LED: PC13<Output<PushPull>> = ();
    static mut MAX_DUTY: f32 = ();
    static mut DEBOUNCER_HANDLER: PortDebouncer<U8, U2> = ();

    #[init(schedule = [debounce])]
    fn init() -> init::LateResources {

        let mut flash = device.FLASH.constrain();
        let mut rcc = device.RCC.constrain();
        let mut afio = device.AFIO.constrain(&mut rcc.apb2);
        let mut gpioa = device.GPIOA.split(&mut rcc.apb2);
        let mut gpiob = device.GPIOB.split(&mut rcc.apb2);
        let mut gpioc = device.GPIOC.split(&mut rcc.apb2);

        let clocks = rcc.cfgr
            .sysclk(64.mhz())
            .pclk1(32.mhz())
            .freeze(&mut flash.acr);

        let mut led = gpioc.pc13.into_push_pull_output_with_state(&mut gpioc.crh, State::High);
        let _ = gpioa.pa1.into_pull_up_input(&mut gpioa.crl);
        let _ = gpioa.pa2.into_pull_up_input(&mut gpioa.crl);
        let _ = gpioa.pa3.into_pull_up_input(&mut gpioa.crl);
        let _ = gpioa.pa4.into_pull_up_input(&mut gpioa.crl);
        let _ = gpioa.pa5.into_pull_up_input(&mut gpioa.crl);

        let c1 = gpioa.pa0.into_alternate_push_pull(&mut gpioa.crl);
        let pwm_hal = device.TIM2.pwm(
            c1,
            &mut afio.mapr,
            20.khz(),
            clocks,
            &mut rcc.apb1,
        );
        let max_duty = pwm_hal.get_max_duty() as f32;

        let debouncer: PortDebouncer<U8, U2> = PortDebouncer::new(40, 200);

        schedule.debounce(Instant::now() + DEBOUNCE_PERIOD.cycles()).unwrap();
        led.set_low();

        init::LateResources {
            LED: led,
            PWM_HANDLER: pwm_hal,
            MAX_DUTY: max_duty,
            DEBOUNCER_HANDLER: debouncer,
        }
    }

    #[idle]
    fn idle() -> ! {
        loop {
            nop();
        }
    }

    #[task(priority = 2, schedule = [debounce], resources = [LED, DEBOUNCER_HANDLER])]
    fn debounce() {
        static mut COUNT: usize = 0;
        schedule.debounce(scheduled + DEBOUNCE_PERIOD.cycles()).unwrap();

        resources.DEBOUNCER_HANDLER.update(!(unsafe {(&(*pac::GPIOA::ptr())).idr.read().bits()}));
        *COUNT += 1;

        if *COUNT == 8 {
            *COUNT = 0;
            let state = resources.DEBOUNCER_HANDLER.get_state(1).unwrap();

            match state {
                BtnState::ChangedToPressed | BtnState::Repeat => {
                    resources.LED.toggle();
                }
                _ => {}
            }
        }
    }
    
    extern "C" {
        fn EXTI1();
    }
};`

`cortex-m-rtfm = {version = "0.4.3", features=["timer-queue"]}`

If I change 
`#[init(schedule = [debounce])]` to `#[init(spawn = [debounce])]` and `schedule.debounce(Instant::now() + DEBOUNCE_PERIOD.cycles()).unwrap();` to `spawn.debounce().unwrap();` everything works as expected.

I will try to test a strip down version later tomorrow, so other people can try to replicate.

> Aldo what's your setup: do you power with a stlink? Plugged to z computer or a power supply?

I kinda tried both ways, with the stlink and powering through the usb from the bluepill (stlink disconnect). But the stlink resets the mcu after flashing, so things work because of that. If there is no reset (just power on) then it takes close to 10 seconds after init (I know because of the led method in the end of it) to start the task.

jmgao, 2019-07-26:
I've been running into this as well. It looks like the value of the DWT CYCCNT register seems to always be 1073741824 (2^30) in init?

Ah, it looks like the problem is that on cold boot, the DWT unit is off, so we schedule tasks relative to 2^30, then we turn the DWT unit on after init, which resets the cycle counter to 0. `1073741824 / (72 MHz)` is ~15 seconds.

I'll upload a pull request shortly.

thalesfragoso, 2019-08-26:
I haven't tested this yet, but it looks like it got solved on https://github.com/japaric/cortex-m-rtfm/pull/221

closed 2019-08-26 by thalesfragoso

dbrgn, 2020-01-13:
It seems that this problem is back on rtfm 0.5.

Reproducer for the blue pill:

```rust
#![no_main]
#![no_std]
#![deny(unsafe_code)]

extern crate panic_halt;

use rtfm::app;
use rtfm::cyccnt::{Instant, U32Ext};
use stm32f1xx_hal::{prelude::*, pac};
use stm32f1xx_hal::gpio::{Output, OpenDrain, State, gpioc};

const POLL_PERIOD: u32 = 48_000 * 50; // ~50ms, for testing purposes

#[app(device = stm32f1::stm32f103, peripherals = true, monotonic = rtfm::cyccnt::CYCCNT)]
const APP: () = {
    struct Resources {
        led: gpioc::PC13<Output<OpenDrain>>,
    }

    #[init(schedule = [toggle_led])]
    fn init(ctx: init::Context) -> init::LateResources {
        // Peripherals
        let mut core: rtfm::Peripherals = ctx.core;
        let device: pac::Peripherals = ctx.device;

        // Get reference to peripherals
        let mut rcc = device.RCC.constrain();
        let mut gpioc = device.GPIOC.split(&mut rcc.apb2);
        let mut flash = device.FLASH.constrain();

        // Initialize (enable) the monotonic timer (CYCCNT)
        core.DCB.enable_trace();
        core.DWT.enable_cycle_counter();

        // Clock configuration
        let _clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(48.mhz())
            .pclk1(24.mhz())
            .freeze(&mut flash.acr);

        // Schedule polling timer
        ctx.schedule.toggle_led(Instant::now()).unwrap();

        // Assign resources
        let led = gpioc.pc13.into_open_drain_output_with_state(&mut gpioc.crh, State::High);
        init::LateResources { led }
    }

    #[task(resources = [led], schedule = [toggle_led])]
    fn toggle_led(ctx: toggle_led::Context) {
        ctx.resources.led.toggle().unwrap();
        ctx.schedule.toggle_led(ctx.scheduled + POLL_PERIOD.cycles()).unwrap();
    }

    // RTFM requires that free interrupts are declared in an extern block when
    // using software tasks; these free interrupts will be used to dispatch the
    // software tasks.
    extern "C" {
        fn SPI1();
        fn SPI2();
    }
};
```

On power reset, the blue pill LED will blink immediately. But after reset (or when flashing firmware), it takes a seemingly random amount of time (1-30s) until it starts blinking.

japaric, 2020-01-21:
@dbrgn `Monotonic::now` is [allowed to return nonsensical values](https://github.com/rtfm-rs/cortex-m-rtfm/blob/1e827e24d024e910b44f6cc6db278e4c66a77683/src/lib.rs#L146-L148) in `init`. `cyccnt::Instant` does return nonsensical values in `init`. 

In this line:

``` rust
ctx.schedule.toggle_led(Instant::now()).unwrap();
```

You should use `ctx.start` or `Monotonic::zero()` in `init`. They are both guaranteed to be return the system "start time".

dbrgn, 2020-01-22:
@japaric ah, you are right! And it seems that [the book shows it correctly as well](https://rtfm.rs/0.5/book/en/by-example/timer-queue.html).

I wonder if a compile warning could be generated if  `Monotonic::now()` is used inside `init`?

perlindgren, 2020-01-23:
I'm not sure we can detect this at compile time, but we could do it at run time. BUT that would probably come with a cost/OH for all cases using `now()`. It might be a price worth to pay, for ensuring defined behavior. Personally I don't like the idea of UD being avoided only on behalf of the user actually reading the manual.

PR #221 (2019-08-20): Enable the DWT unit before poking at its registers. (v0.4.x)
On cold boot, the DWT unit is off, which means our attempts to clear and
disable the cycle counter in pre-init don't work. This seems to result
in access to CYCCNT always returning 1073741824, which delays tasks
scheduled in init by that many cycles (~15s at 72 MHz).

Fixes #196.
commit ec1d5080 (2019-10-23): Enable the DWT unit before poking at its registers.

On cold boot, the DWT unit is off, which means our attempts to clear and
disable the cycle counter in pre-init don't work. This seems to result
in access to CYCCNT always returning 1073741824, which delays tasks
scheduled in init by that many cycles (~15s at 72 MHz).

Fixes #196.