rtic-rs/rtic #197: Best way to configure interrupt handlers

little-arhat, 2019-06-11:
Hi,

My app has several diferent configurations (set of pins, constants, etc) that I control via cargo features and conditional compilation. I've tried to do the same for interrupt handlers with `cfg_attr` but it didn't work:

```lang:rust

    #[cfg_attr(configuration = "configuration_dev",
               interrupt(binds=EXTI0,
                         resources = [EXTIH, AHRS, LOG, DEBUG_PIN, TELE]))]
    #[cfg_attr(configuration = "configuration_prod",
               interrupt(binds=EXTI15_10,
                         resources = [EXTIH, AHRS, LOG, DEBUG_PIN, TELE]))]
    fn handle_int(ctx: handle_int::Context) {
        let _ = ctx.resources.DEBUG_PIN.set_high();
        ...
    }
```

```
error: this item must live outside the `#[app]` module
   --> src/main.rs:138:5
    |
138 |     #[cfg_attr(configuration = "configuration_dev",
    |     ^

error: aborting due to previous error
```

Then I've tried to create two interrupt handlers with identical bodies (with the idea to abstract that into free function), but that didn't work either: as I have no way to prove to `rtfm-macros` those two interrupt handlers can't run at the same, it generates code with locks (I guess), so I get errors like:

```
error[E0596]: cannot borrow `ctx.resources.DEBUG_PIN` as mutable, as `ctx` is not declared as mutable
   --> src/main.rs:141:17
    |
140 |     fn handle_mpu_0(ctx: handle_mpu_0::Context) {
    |                     --- help: consider changing this to be mutable: `mut ctx`
141 |         let _ = ctx.resources.DEBUG_PIN.set_high();
    |                 ^^^^^^^^^^^^^^^^^^^^^^^ cannot borrow as mutable

error[E0596]: cannot borrow `ctx.resources.TELE` as mutable, as `ctx` is not declared as mutable
   --> src/main.rs:144:30
    |
140 |     fn handle_int_0(ctx: handle_int_0::Context) {
    |                     --- help: consider changing this to be mutable: `mut ctx`
...
144 |         let mut maybe_tele = ctx.resources.TELE.take();
    |                              ^^^^^^^^^^^^^^^^^^ cannot borrow as mutable

```

What is the best way for conditional compilation of interrupt handers then? 

Thanks.

little-arhat, 2019-06-12:
@japaric is it possible to implement smth like https://github.com/japaric/cortex-m-rtfm/issues/110 but for tasks/interrupts? 

Or maybe do smth like this

```lang:rust

#interrupt[binds={fea1 => INT1, fea2 => INT2}]
fn foo() {}

```

to perform analysis, and then forward `#[cfg(fea1)]` to `fn INT1`, etc.

little-arhat, 2019-06-12:
Made it to work by an ugly hack:

```lang:rust

#[app(device = hal::pac)]
const APP: () = {
    // skip ..

    #[interrupt(binds=EXTI15_10,
                resources = [EXTIH, AHRS, LOG, DEBUG_PIN, TELE])]
    fn handle_mpu_prod(ctx: handle_mpu_prod::Context) {
        #[cfg(configuration = "configuration_prod")]
        handle_mpu(ctx);
    }

    #[interrupt(binds=EXTI0,
                resources = [EXTIH, AHRS, LOG, DEBUG_PIN, TELE])]
    fn handle_mpu_dev(ctx: handle_mpu_dev::Context) {
        #[cfg(configuration = "configuration_dev")]
        handle_mpu(ctx);
    }
}

#[cfg(configuration = "configuration_prod")]
type CtxType<'a> = handle_mpu_prod::Context<'a>;
#[cfg(configuration = "configuration_dev")]
type CtxType<'a> = handle_mpu_dev::Context<'a>;
fn handle_mpu(mut ctx: CtxType) {
    let _ = ctx.resources.DEBUG_PIN.set_high();
   // skip ..
}
```

but this solution is error prone, so I think it would be nice if `rtfm` provided a way to configure that.

japaric, 2019-06-24:
> is it possible to implement smth like #110 but for tasks/interrupts?

You can `cfg` resources and entire software tasks.

``` rust
#[cfg(something)]
static mut X: .. = ..;

#[cfg(something_else)]
#[task(..)]
fn a(..) { .. }
```

If you want to change any of the arguments of a `#[task]`, like the `resources` list or its `priority`, then you have to `cfg` the whole `rtfm::app` spec.

``` rust
#[cfg(something)]
#[rtfm::app(..)]
const APP: () = {
    #[task(priority = 2, resources = [A, B])]
    fn a(..) { .. }

    // ..
};

#[cfg(not(something))]
#[rtfm::app(..)]
const APP: () = {
    #[task(priority = 3, resources = [A, B, C])]
    fn a(..) { .. }

    // ..
};
```

This is necessary because the ceiling analysis is not aware of `cfg`s.

If the only thing you want to change is which interrupts a task binds to you can write your own `device` module.

``` rust
// NOTE untested
mod my_bindings {
    use bare_metal::Nr;

    pub use pac::Peripherals;
    pub use pac::NVIC_PRIO_BITS;

    // maps interrupt names to their "numbers" (position in vector table)
    #[repr(u8)]
    #[derive(Clone, Copy)]
    pub enum Interrupt {
        #[cfg(something)]
        MY_EXTI = 10,
        #[cfg(not(something))]
        MY_EXTI = 8,

        // .. all other interrupts that you'll use ..
    }

    impl Nr for Interrupt {
        fn nr(&self) -> u8 {
            *self as u8
        }
    }
}

// NOTE: `my_bindings` instead of `pac`
#[rtfm::app(device = my_bindings)]
const APP: () = {
    #[interrupt(binds = MY_EXTI)]
    fn a(..) { .. }
};
```

However, note that what RTFM expects of the `device` module is currently only specified as "must be a `svd2rust` generated crate" so what you see above may not be enough in a new patch release --  that is your app crate could break in a new patch release. We'll likely fully specify the contents of `device` module in the next minor version, though, as this is required for multi-core support.

> Or maybe do smth like this

I don't think I personally want to go down this route as people might expect being able to use `cfg`s in all arguments, which would make the ceiling analysis rather difficult to implement and test for correctness.

little-arhat, 2019-06-24:
@japaric Thanks, proviging custom device looks like reasonable solutions (as I can put [other configuration](https://github.com/copterust/fcfs-rtfm/blob/master/src/boards.rs#L37) there as well). I think it would be nice to document these patterns somewhere.

little-arhat, 2019-10-23:
Although, using custom interrupt names breaks interrupt handling, as rtfm will translate
```
#[task(binds=MY_EXTI)]
fn handle()
```

into 

```
#[no_mangle]
fn MY_EXTI() {
}
```

And then interrupt will be handled by default handler instead.

korken89, 2019-12-29:
This discussion seems finished.

little-arhat, 2019-12-29:
There is still no recommended solution, though.

korken89, 2019-12-29:
Okey, reopening

little-arhat, 2021-02-18:
I tried to use `cfg_attr`, like this:

```
   #[cfg_attr(configuration = "conf2",
               task(binds=EXTI15_10,
                    resources = [...]))]
    #[cfg_attr(configuration = "conf1",
               task(binds=EXTI0,
                    resources = [...]))]
    fn handle_mpu(ctx: handle_mpu::Context) {
      // code
    }
```

But then I can't even look and the `cargo expand` output:

```
error: this item must live outside the `#[app]` module
   --> src/main.rs:244:8
    |
244 |     fn handle_mpu(ctx: handle_mpu::Context) {
    |        ^^^^^^^^^^^^^^^^
```

So I have to declare two tasks and and use `#[cfg]` to leave one of them empty. This, however, prevents me from usin `task_locals`, because, well, rtic is not able to check function bodies, I guess. 

Is there a way to define only one task depending on configuration? Is that anywhere in the roadmap? Maybe someone can give a pointers/direction on how to implement cfg support?

korken89, 2021-02-18:
The issue with cfg support is that the order of attribute evaluation in rust is undefined, this causes large problems with the analysis because we cannot know which is active. So for now the user needs to work around until this is fixed in proc macros.

little-arhat, 2021-02-18:
Is there open issue I can watch to track proc-macro fix?

Also, I guess, it should be possible to make rtic-syntax cfg aware?

little-arhat, 2021-02-18:
@korken89 I don't think attribute evaluation plays a role here. When we encounter #[task] we are inside rtic-app macro, so it's up to rtic to decide how to process token stream. 

I don't think rust can go inside an item first to expand cfg, and then invoke rtic-app macro. 

For codegen it would be enough to add `cfg` argument to `task` attribute and then forward it to all generated items, but I'm not sure how to modify analysis part to account for `cfg`.

perlindgren, 2021-02-19:
Attribute evalution plays a key role here, as far as I know we cannot evaluate the `cfg`s at compile time (inside the proc-macro), thus we cannot take the cfgs into account for the analysis.

little-arhat, 2021-02-19:
@perlindgren but if only binds are configurable, analysis should be fine, right?

perlindgren, 2021-02-20:
Yes, in some special cases, where the analysis and codegen does not depend on the evaluation of the attribute. But its hard to foresee what kind of problems might arise due to cfgs in the general case. Introducing additional syntax in RTIC to make some cfgs possible might work, but I rather wait and hope that we can find a better way.

little-arhat, 2021-02-23:
Btw, I patched `rtic-syntax` and `cortex-m-rtic` to support configurable binds:

```
    #[task(binds=[("configuration_prod", EXTI15_10),
                  ("configuration_dev", EXTI0)],
           resources = [extih, ahrs, log, debug_pin,
                        channel, control, state, motors])]
    fn handle_mpu(mut ctx: handle_mpu::Context) {
    }
```

code above will generate something like:

```
    #[allow(non_snake_case)]
    #[no_mangle]
    #[cfg(configuration = "configuration_dev")]
    unsafe fn EXTI0() {
        const PRIORITY: u8 = 1u8;
        rtic::export::run(PRIORITY, || {
            crate::app::handle_mpu(handle_mpu::Context::new(&rtic::export::Priority::new(
                PRIORITY,
            )))
        });
    }

    #[allow(non_snake_case)]
    #[no_mangle]
    #[cfg(configuration = "configuration_prod")]
    unsafe fn EXTI15_10() {
        const PRIORITY: u8 = 1u8;
        rtic::export::run(PRIORITY, || {
            crate::app::handle_mpu(handle_mpu::Context::new(&rtic::export::Priority::new(
                PRIORITY,
            )))
        });
    }
```

(and similar bits, in pre-init).

Current syntax is also supported and no `cfg` guards are generated in this case. 

https://github.com/copterust/rtic-syntax/tree/feature-configurable-binds

https://github.com/copterust/cortex-m-rtic/tree/feature-configurable-binds

wbuck, 2021-07-02:
So I also just ran into this same issue. I'm trying to conditionally swap resources/interrupts based on features. Judging from the above conversation this is not possible with RTIC. Is that correct?

romixlab, 2021-07-13:
+1
This would be a great feature to have. Simplest case: different revisions of the board might require some pin shuffling, so interrupt handlers change.

franeklubi, 2022-03-24:
Has any work on this been done? I need a way to create tasks based on features

perlindgren, 2022-03-24:
No decision taken yet.

Would the approach suggested by @little-arhat be sufficient to support you use case @franeklubi ?
/Per

franeklubi, 2022-04-04:
I've achieved my desired results by:
1. `cfg`ing whole task body if needed
2. Not configuring interrupts for particular tasks
3. Not spawning particular tasks
4. Since you can't `cfg`-out resources, because it complains the type does not exist - instead of importing - creating a type alias to unit `()`, effectively making the resource unusable.

Hope that may help someone!

little-arhat, 2022-05-16:
Hello! 

@perlindgren, what is the current stage of this, post v1? 

Should I attempt to update my patch for v1 and submit PR?

perlindgren, 2022-05-16:
Hi. 

Its been a while since looking at this, the point would be just `cfg` to bind different handlers, not making tasks/resourrces optional or?

little-arhat, 2022-05-31:
@perlindgren the idea is to have task bound to different interrupts conditionally:

https://github.com/rtic-rs/cortex-m-rtic/issues/197#issue-454672801 — example problem
https://github.com/rtic-rs/cortex-m-rtic/issues/197#issuecomment-784546762 — example solution

cfg-whole-task doesn't work:

```
//! examples/cfg-whole-task.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use panic_semihosting as _;

#[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
mod app {
    use cortex_m_semihosting::debug;

    #[shared]
    struct Shared {
        count: u32,
        #[cfg(never)]
        unused: u32,
    }

    #[local]
    struct Local {}

    #[init]
    fn init(_: init::Context) -> (Shared, Local, init::Monotonics) {
        (
            Shared {
                count: 0,
            },
            Local {},
            init::Monotonics(),
        )
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        debug::exit(debug::EXIT_SUCCESS); // Exit QEMU simulator

        loop {
            cortex_m::asm::nop();
        }
    }

    #[cfg(configuration = "configuration_dev")]
    #[task(binds = GPIOA, shared = [count])]
    fn foo_a(mut _cx: foo_a::Context) {
        _cx.shared.count.lock(|count| *count += 10);
    }

    #[cfg(configuration = "configuration_prod")]
    #[task(binds = GPIOB, shared = [count])]
    fn foo_b(mut _cx: foo_b::Context) {
        _cx.shared.count.lock(|count| *count += 10);
    }
}

```

```
error[E0412]: cannot find type `Context` in module `foo_a`
  --> examples/cfg-whole-task.rs:10:1
   |
10 | #[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `foo_a`
   |
   = note: this error originates in the attribute macro `rtic::app` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing one of these items
   |
12 |     use core::task::Context;
   |
12 |     use crate::app::idle::Context;
   |
12 |     use crate::app::init::Context;
   |
help: if you import `Context`, refer to it directly
   |
10 - #[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
   |

error[E0412]: cannot find type `Context` in module `foo_b`
  --> examples/cfg-whole-task.rs:10:1
   |
10 | #[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `foo_b`
   |
   = note: this error originates in the attribute macro `rtic::app` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing one of these items
   |
12 |     use core::task::Context;
   |
12 |     use crate::app::idle::Context;
   |
12 |     use crate::app::init::Context;
   |
help: if you import `Context`, refer to it directly
   |
10 - #[rtic::app(device = lm3s6965,  dispatchers = [SSI0, QEI0])]
```

little-arhat, 2022-10-20:
@perlindgren ^ any comments on the above?

little-arhat, 2023-07-01:
Hello! @AfoHT , @korken89  — any recommendations how to solve above issue with rtic 1 or 2? Thank you!

Ddystopia, 2025-01-04:
Maybe rtic or rtfm might support interrupt aliases? That would'be been a good solution imo.