rtic-rs/rtic #198: [RFC] don't auto-wrap mutable references to resources in the `Exclusive` newtype.

japaric, 2019-06-14:
## Current state

Resources (`static mut` variables) accessed from high priority tasks appear in
the `Resources` struct as mutable references wrapped in the `Exclusive` newtype.

``` rust
// defined in cortex-m-rtfm
pub mod rtfm {
    pub struct Exclusive<'a, T>(pub &'a mut T);

    impl DerefMut for Exclusive { /* type Target = T */ }
}

#[rtfm::app(device = ..)]
const APP: () = {
    static mut X: u32 = 0;

    #[task(resources = [X], priority = 1)]
    fn foo(c: foo::Context) {
        // ..
    }

    #[task(resources = [X], priority = 2)]
    fn bar(c: bar::Context) {
        let mut x: rtfm::Exclusive<u32> = c.resources.X;

        // direct access
        *x += 1;

        // access through a lock
        x.lock(|x| *x += 1);
    }

    // ..
};
```

The motivation for this behavior is letting the user access these resources both
with the `lock` API and without it (as shown in the previous example).

## Proposal

Stop auto-wrapping these references in the `Exclusive` newtype. If the end user
needs the `Exclusive` newtype they can wrap the reference themselves -- the
newtype only field is public so `Exclusive(resources.foo)` will create an
`Exclusive` value.

## Rationale

In my opinion this auto-wrap feature has very little actual use and it's not
worth the implementation complexity. You see not all resources appear behind an
`Exclusive` newtype when it's not necessary to use the `lock` API.

``` rust
#[rtfm::app(device = ..)]
const APP: () = {
   static mut Y: u32 = 0;

    // only this task access the resource `Y`
    #[task(resources = [Y])]
    fn baz(c: baz::Context) {
        // no `Exclusive` newtype here
        let y: &mut u32 = c.resources.Y;
    }
};
```

End users are more likely to hit compilation errors due to this inconsistency
rather than to actually reach out for the `Exclusive` newtype.

Here are some of the observable differences:

``` rust
#[rtfm::app(device = ..)]
const APP: () = {
   static mut X: u32 = 0;
   static mut Y: u32 = 0;
   static Z: u32 = 0;

    #[task(resources = [X], priority = 1)]
    fn foo(c: foo::Context) {
        // ..
    }

    #[task(resources = [X], priority = 2)]
    fn bar(mut c: bar::Context) {
        // ^^^ (a) needs to be `mut c` or the next statement won't work
        *c.resources.X += 1;

        // (b) needs a borrow
        quux(&mut c.resources.X);

        // (c) needs a borrow
        quux_ref(&c.resources.X);

        // (d) needs a deref
        println!("{}", *c.resources.X);
    }

    // only this task access the resource `Y`
    #[task(resources = [Y, Z])]
    fn baz(c: baz::Context) {
        // (a) doesn't need `mut c`
        *c.resources.Y += 1;

        // (b) doesn't need a borrow
        quux(c.resources.Y);

        // (c) doesn't need a borrow
        quux_ref(c.resources.Y);

        // like Y this doesn't need a borrow
        quux_ref(c.resources.Z);

        // (d) doesn't need a deref
        println!("{}", c.resources.Y);
    }
};

fn quux(_: &mut u32) {}
fn quux_ref(_: &u32) {}
```

korken89, 2019-06-25:
I agree fully with you here @japaric

korken89, 2019-06-25:
On a side note, why is this even allowed?
IMO we should always use the `.lock( .. )` syntax, as now we can simply change priorities and the compilation may fail making the specification fragile to this "ease of use".

Maybe this should be in another discussion.

japaric, 2019-06-29:
> I agree fully with you here @japaric

OK then let's FCP this with disposition to merge

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi 

---

> On a side note, why is this even allowed?

Because it's more convenient that `lock`-ing things that don't need a lock, which increases rightward drift and gets worse once you have more resources (nesting). I personally don't find `lock`s that common (that is resources shared at different priorities) and find even less common the situation where I change task priorities midway development -- more often than not I use late resources as runtime initialized task local data, that is late resources owned by a single task. But that's just my impression as someone who mostly writes compilers rather than actual embedded applications :-).

> IMO we should always use the .lock( .. ) syntax

I would agree if we had automatic selection of task priorities (based on timing information and whatnot).

> Maybe this should be in another discussion.

It'd be best to have another issue for this discussion.

TeXitoi, 2019-06-29:
OK for me.

korken89, 2019-07-02:
Thanks for the feedback! All ok from my side, I will mull this discussion over a bit and if I find good motivations I'll start another issue.

japaric, 2019-07-08:
:tada:  This RFC has been formally approved. Implementation is in PR #205 (I'm going to keep this open until the PR lands)

japaric, 2019-09-15:
Done in PR #205

closed 2019-09-15 by japaric