rtic-rs/rtic #200: [RFC] user defined `monotonic` timer

japaric, 2019-06-14:
## Current behavior

The `schedule` API internally makes uses of two timers: the `DWT_CYCCNT` (the
cycle counter) as a monotonic timer / counter and the `SysTick` (system timer)
to generate timeouts. Currently, these can't be changed and the result is that
the `schedule` API can't be used on ARMv6-M where `DWT_CYCCNT` doesn't exist.

## Proposal

Require that the user specifies the monotonic timer that will be used to
implement the `schedule` API. The user will still be able to use `DWT_CYCCNT` as
the monotonic timer but this will *not* be the default; the user must specify
this timer, or some other timer, in their application.

This RFC does *not* propose a mechanism to sub the `SysTick` timer with a
different timer.

## Rationale

The `DWT_CYCCNT` is not an appropriate monotonic timer for multi-core
applications: each Cortex-M core has its own `DWT` peripheral and its own cycle
counter so it's not possible to synchronize these two counters (there's no
register to do this), plus the counters may be operating at different frequencies
resulting in one core's `Instant` having a different meaning that other cores'
`Instant`s. Also, as was previously mentioned the `DWT_CYCCNT` is not available
on ARMv6-M cores; this limits on which devices one can use the `schedule` API.

In multi-core applications it's better to use a device specific, constant
frequency timer visible to all cores as the monotonic timer. And in single-core
applications one may want to use a 64-bit timer or a prescaled 32-bit timer as
the monotonic timer; this way one can `schedule` tasks with long periods, e.g.
in the order of seconds (cc @jamesmunns).

Lastly, using a device-specific timer as the monotonic timer lets programmers
use the `schedule` API on single-core ARMv6-M devices.

## Detailed design

### The `Monotonic` trait

The following trait will be added to the `cortex-m-rtfm` crate.

``` rust
/// A monotonic clock / counter
pub unsafe trait Monotonic {
    /// A measurement of this clock
    type Instant: Copy + Ord + Sub;

    /// The ratio between the SysTick (system timer) frequency and the frequency of this clock
    fn ratio() -> u32;

    /// Returns the current time
    ///
    /// NOTE this must be an atomic operation
    fn now() -> Self::Instant;

    /// Resets the counter to *zero*
    ///
    /// NOTE the runtime will execute this method with interrupts disabled
    unsafe fn reset();

    /// A `Self::Instant` that represents a count of *zero*
    fn zero() -> Self::Instant;
}
```

This trait represents a monotonic timer. The trait is meant to be implemented on
global singleton ZSTs.

Also, although it's not shown in the `trait` interface, the RTFM runtime expects
[... 240 lines cut ...]
        // (yes, this is fine to do before `some_timer` has been initialized)
        c.schedule.foo(c.start + Duration::from_secs(1).into());

        // initialize `some_timer` (e.g. set prescaler)

        // ..
    }
};
```

## Unresolved questions

- Should we place the `Monotonic` trait in a separate crate (e.g.
  `cortex-m-rtfm-traits`) so that HAL authors can implement it without pulling
  the whole `cortex-m-rtfm` crate, which has lots of dependencies?

- Should we default to the `CYCCNT` implementation when no `monotonic` argument
  is given and the `schedule` API is used? But note that this will not work when
  the target is ARMv6-M and we won't be able to provide a good error message
  because it's not possible to know the compilation target at macro expansion
  time.

---

cc #115

jonas-schievink, 2019-06-14:
Can you explain why `Monotonic` needs to be an `unsafe` trait, and why its `reset` function is `unsafe` as well? What are the safety constraints on implementing `Monotonic` and calling `reset`?

japaric, 2019-06-19:
@jonas-schievink Hmm, actually I don't think the trait needs to be `unsafe`; I can't see how it could break memory safety even if it's implemented in a nonsensical way. 

`reset` is `unsafe` because "it must be called within a critical section and it must be called at most once", the implementer can rely on these conditions being held when the runtime invokes the function. All the other methods are safe because their implementations must be sound even when invoked concurrently and from different cores.

japaric, 2019-06-24:
I have made two amendments to the RFC:

- `Monotonic` is no longer a `unsafe` trait

- a `MultiCore` marker trait has been added. The DSL will check at compile time that the `monotonic` argument implements this trait when the `schedule` API is used in more than one core. The `CYCCNT` monotonic timer will *not* implement this marker trait -- the `CYCCNT` doesn't behave correctly in multi-core context.

korken89, 2019-06-25:
I quite like this. 
However I think `ratio` should return a fraction to better support clock differences.

Looking into a traits crate could be a good option. Are there other traits we'd like to add?

When it comes to a default timer, I'd leave it explicit to easier find issues when `CYCCNT` does not exist (rather that than cryptic error messages).

japaric, 2019-06-29:
> However I think ratio should return a fraction to better support clock differences.

Yes, good idea. Do you think a tuple `(u32, u32)` would do or should we use a `struct Fraction { num: u32, den: u32 }`?

> Are there other traits we'd like to add?

Likely one more that lets you sub the SysTick with a user defined timeout timer. 

There's also the `Mutex` trait to write generic code that uses resources but that one will go in the `rtfm-core` crate (as per RFC #203).

Instead of creating a `cortex-m-rtfm-traits crate` we could directly add these traits to `rtfm-core`. The `MultiCore` marker trait def sounds like it should go in `rtfm-core`. The `Monotonic` one, I'm not sure; may be best to wait and see what other RTFM ports use to implement the `schedule` API.

---

@TeXitoi think we can move forward with this?

(I'm going to set up the tickbox thing in any case)

Let's FCP merging this:

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi 

Note that we still need to decide on the return type for the `ratio` method.

TeXitoi, 2019-06-29:
OK for me.

korken89, 2019-07-02:
I think a `struct Fraction { num: u32, den: u32 }`, people from C++ will feel more at home as well then.
All ok from my side!

japaric, 2019-07-08:
:tada:  This RFC, with the `Fraction` amendment, has been formally approved . Implementation is in PR #205 (I'm going to keep this open until the PR lands)

perlindgren, 2019-08-01:
In the original work on RTFM a generic timer implementation was discussed, and initial experiments were conducted to verify feasibility.  http://ceur-ws.org/Vol-1464/ewili15_16.pdf

There is discussion on using multiple timers/timer queues to reduce priority inversion. In short, a postponed task with a priority of x, will cause priority inversion if (put in a queue) handled by a timer with priority higher (more important) than x. So ideally we should associate a timer for each priority level that postponed task hold (in order to minimize priority inversion).  

One possible approach is to give RTFM a set of "free" timers (implementing the "dispatch" trait). (We have already a similar approach to allowing RTFM to know about available interrupt handlers for the "soft" tasks, the somewhat "ugly" extern C.) 

If the number of available timers is NOT sufficient (more priorities of postponed tasks than available timers), we can think about some heuristics for allocation (by default) or guided by annotations (in the free-timer list the user could say that he wants to associate the timer with queue handled at a particular priority). A sensible heuristic might be to distribute the timers top down (priority vice), so at least the highest priority tasks won't suffer from the priority inversion. 

Ultimately, for a hard real-time system it boils down to response time analysis (and overall schedulability). Priority inversion (due to the timer queue handling/dispatch) is one piece of the puzzle. To that end, it makes full sense to introduce the timers to the RTFM model, and perform the system wide analysis on the complete model (including the timer tasks). The message queues becomes just shared resources (here lock free implementations are of great help reducing blocking, so we are in a particular good spot with RTFM).

Well why bother, RTFM works fine as is, and my application does not have any hard real time constraints.

While this is in general true, RTFM is indeed the most efficient real-time scheduler out there, and your application logic may not have any explicit timing constraints, hard constraints typically trickle in from the interaction with the underlying hardware (hidden by HAL/drivers etc.). E.g., we assume the interrupt handler (task) for the UART, etc. to handle communication without overflowing the input buffer. If that task is exposed to excessive interference by the message passing mechanism, the task misses its deadline and data is lost. We can prevent that by a "multiple timer" message passing.

In the Rust RTFM re-implementation, priority assignment is currently manual, in the original RTFM model priorities were assigned based on task deadline information (which allows reasoning on timing/response times etc.). For I/O interaction those deadlines can be derived from the inter-arrival of events (e.g., in the case of a UART, it would relate to the baud rate). A bit out of topic perhaps, but wouldn't it be great with a bit more of const evaluation of init, and having deadlines derived automatically by the HAL/driver implementations???? I fear however that Rust is not yet powerful enough to that type of const evaluation during proc-macro execution, but perhaps an extension to HAL could be possible with declarative style init (accessible to the proc-macro ... just dreaming.)

In the short term, lets focus on generic timers, and a reasonable heuristic for timer allocations.
/Per

japaric, 2019-09-15:
Done in PR #205

closed 2019-09-15 by japaric