rtic-rs/rtic #204: [RFC] heterogeneous multi-core support

japaric, 2019-06-14:
## Summary

Add *experimental* heterogeneous multi-core support behind a Cargo feature.

## Detailed design

### Semantics

In multi-core applications each core will basically run its own independent RTFM
sub-application. Each core will have its own tasks and resources (`static mut`
variables) and the cores will be able to communicate using message passing:
cross-core `spawn` & `schedule` API.

It's not possible to share resources (`static mut` variables) between cores.
However, it is possible for one core to initialize another core's resources and
it's also possible to share `static` variables between cores.

### Feature gating

The heterogeneous multi-core support will live behind a `heterogeneous` Cargo
feature considered *experimental*. We'll reserve the right to do breaking
changes to this API in patch releases and will recommend end users to pin
to a exact version of `cortex-m-rtfm` when using this API (i.e. `=0.5.0`
in `Cargo.toml`)

### Syntax

The `#[app]` attribute will gain a `cores` argument that indicates the number of
cores the application will use. The argument expects an integer value equal
or greater than 2. Omitting this argument indicates that the application is a
single-core application.

``` rust
// dual-core application
#[rtfm::app(cores = 2)]
const APP: () = {
   // ..
};
```

In multi-core mode all tasks, `#[init]` and `#[idle]` functions need to indicate
in which core they'll run using the `core` argument.

``` rust
// dual-core application
#[rtfm::app(cores = 2)]
const APP: () = {
    // core #0 initialization function
    #[init(core = 0)]
    fn init(c: init::Context) {
        // cross-core message passing
        c.spawn.ping(0).ok();
    }

    #[task(core = 0)]
    fn pong(c: pong::Context, x: u32) {
        c.spawn.ping(x + 1).ok();
    }

    // core #1 initialization function
    #[init(core = 1)]
    fn init(_: init::Context) {
        // ..
    }

    #[task(core = 1)]
    fn ping(c: ping::Context, x: u32) {
        c.spawn.pong(x + 1).ok();
    }
};
[... 316 lines cut ...]

## Unresolved questions

Should we allow more than one core bind a hardware task to the same interrupt?
This seems like it should be supported by the hardware: the semantics would be
that both cores will start executing some hardware task when *a peripheral*
fires the corresponding interrupt signal. Note that if one core executes
`rtfm::pend(SomeInterrupt)` this will have *no* effect on other cores.

``` rust
#[rtfm::app(cores = 2)]
const APP: () = {
    // both foo and bar will start and run (in parallel) when the CTIMER0 times out

    #[interrupt(core = 0, binds = CTIMER0)]
    fn foo(_: foo::Context) {
        // ..
    }

    #[interrupt(core = 1, binds = CTIMER0)]
    fn bar(_: bar::Context) {
        // ..
    }
};
```

eddyp, 2019-06-15:
@japaric Regarding the unanwered question, it is possible that the same interrupt number have different meanings on the same SoC, especially if one core is M class and another is A class (and on armv8, there needs to be some duplication, depending on the EL level.

japaric, 2019-06-18:
I have revised the RFC to include *homogeneous* multi-core support and have better documented the contract with the {device} crate. Examples of the homogeneous multi-core support can be found in the [lpcxpresso55S69](https://github.com/japaric/lpcxpresso55S69/tree/master/lpc55s6x/examples) repository, though the examples are unexciting because the API is still the same.

I have also added a new unresolved question about compile time checking uses of the `xpend` API from the application.

---

@eddyp There's an `Interrupt` enumeration per core that maps the interrupt name to its interrupt number. The runtime will use the appropriate enumeration when signaling interrupts and when configuring the interrupts in the NVIC so I don't think that would be an issue.

korken89, 2019-06-25:
Overall I am very positive to this as I was not sure how SRP would handle the multi-core support!
Are there (cheap) development boards to test this on? This is the kind of feature that needs a lot of evaluation IMO :)

japaric, 2019-06-29:
I think we can go ahead and FCP merge this. There isn't much choice in term of syntax and the biggest piece of design work before this can be called stable is going to be the contract with the `device` crate which is probably going to see quite a bit of iteration (see RFC #211 already) but the RFC let's iterate that part in patch releases so I think we are good on that front.

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi 

---

> Are there (cheap) development boards to test this on?

I have been using the lpcxpresso54114 (+pyocd) in heterogeneous mode and lpcxpresso55S69 (+jlink) in homogeneous mode (though this device is asymmetric the first core has an FPU so one should use a different compilation target for that one) with great success. The support for debugging (GDB) the second core is rather bad, IME, so you have to be a bit creative to figure out what the second core is doing :-). 

I also tried the nucleo WB55 but that was a waste because the second core comes with locked down firmware (read / write protection is enabled) which seems like it going to be pain in the neck to overwrite with any firmware other than ST's signed blobs. (plus this device has a radio with almost zero documentation about it; just 2 pages in the reference manual that include a block diagram of the radio and *no* register map)

TeXitoi, 2019-06-29:
OK for me

korken89, 2019-07-02:
Cool, I will be playing a bit with this!
Maybe the debug tool @therealprof and someone more was working on is a good candidate here for looking into tooling?

japaric, 2019-07-08:
:tada:  This RFC has been formally approved. Implementation is in PR #205 (I'm going to keep this open until the PR lands).

japaric, 2019-09-15:
Done in PR #205

closed 2019-09-15 by japaric