rtic-rs/rtic #210: Timer Queue without Systick

tpwrules, 2019-06-25:
I am experimenting with RTFM on the nrf52840 and am having some issues with the timer queue because of the chip's unconventional Systick implementation.

According to page 19 of the [nrf52840 Product Specification v1.1](https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.1.pdf), the Systick does not count when the CPU is stopped, i.e. during WFI, WFE, or SLEEPONEXIT. This means that the idle function has to be an empty loop if otherwise not used, and cannot go to sleep. This somewhat limits the value of the timer queue since the application cannot turn off the CPU and save power when there are no timer events ready.

I expect the timer queue to remain using systick by default, but I wonder if there is room for some kind of interface to use another timer when advantageous for a particular target. Nordic suggests using the RTC timer instead, or a TIMER. I don't know what other platforms this might apply to.

korken89, 2019-06-25:
That is a nasty one. 
Maybe a solution is to make it so the WFI instruction is only used when the TQ is empty?
What do you think @japaric ?

jonas-schievink, 2019-06-25:
@tpwrules https://github.com/japaric/cortex-m-rtfm/issues/115 would allow plugging in your own timer implementation