rtic-rs/rtic #211: [RFC] [multi-core] `#[shared]` and placement of code and data

japaric, 2019-06-29:
This RFC only affects the multi-core modes proposed in RFC #204.

## Background

Multi-core Cortex-M devices usually have several memory regions, each one
connected to a different bus (AHB port in ARM Cortex-M terms) with the goal of
reducing memory contention. Different cores can access different memory regions
without contention and with predictable performance. It is when two, or more,
cores try to access the same memory region that contention occurs and one core
is given priority over the other, resulting in perceived memory access delay on
the second core.

To keep performance predictable it is important that applications carefully
place their resources (code and data) in a way that minimizes contention. To
give an example of the important of memory placement: consider the following
`homogeneous` multi-core RTFM application running on the LPC55S69, a device with
two Cortex-M33 cores, one Flash bank and 5 RAM regions.

``` rust
#[rtfm::app(cores = 2, device = lpc55s6x)]
const APP: () = {
   #[task(core = 0)]
   fn a0(cx: a0::Context) {
       asm::bkpt();
   }

   #[init(core = 1, spawn = [a0])]
   fn init(cx: init::Context) {
       cx.spawn.a0();
   }

   #[cfg(contention)]
   #[idle(core = 1)]
   fn idle(cx: idle::Context) -> ! {
       loop {
           asm::nop();
       }
   }
};
```

If *all* code is placed in Flash then the response time of task `a0` -- in this
case measured as the time it takes to go from interrupt entry to the breakpoint
-- varies depending on whether the second core is doing any work or
sleeping because the second core also loads its instructions from Flash.

Without contention (when `cfg(contention)` evaluates to `false`) the response
time is `26` clock cycles; with contention the response time is `31` clock
cycles -- 20% slower. These numbers correspond to a configuration where the
first core is given *higher* access priority to the Flash.

## Proposal

Specify how the framework places function and data in memory, but in a way that
end users can minimize memory contention in their application.

## Detailed design

### Placement of resources and functions

This RFC proposes that we *specify* the location of functions and static
variables as follows:

All functions and `static mut` variables that *need* to be shared between
the cores will be placed in *shared memory*; everything else will be placed in
memory *local* to the core that uses it.

Or in other words the default is to place items in core-local memory.

Examples of items placed in *local memory*:
[... 265 lines cut ...]
```

`foo` will be placed in Flash because that's the default for this mode. If `foo`
is *not* inlined into `i1` then core #1 will run some code off Flash causing
memory contention with core #0, which runs all its code from Flash.

There are some ways around this issue like placing the shared code (`.text`) in
`SRAM3` to at least never cause contention on Flash, or to use instruction
caches for shared code, if the device has one. Though, the best way to solve the
issue may be to use the `heterogeneous` multi-core mode, even if the device is
homogeneous, as this mode doesn't allow sharing of code, only of data.

---

cc @disasm this may be of interest to you. Out of curiosity, are
(instruction) caches mandatory on SMP RISCV devices? Do devices like the K210
usually have caches?

Not familiar with (instruction) caches on Cortex-M devices (I think that in the
ARMv7-M line only Cortex-M7 devices have them -- dunno if ARMv8-M devices have
them) but it seems to me that with the linker script from the "Homogeneous"
section one could (read-only) cache the whole `.text` section, which contains
all shared code, (and probably also the `.rodata` section) on the second core
(only) to prevent all contention on Flash memory -- at least in the case of the
LPC55S69.

Disasm, 2019-06-30:
@japaric As far as I can see, RISC-V specs tell only what to do if you have caches, but do not require their presence. However, in real-world devices both I$ and D$ caches are present: K210, FU540 and even FE310.
That's why this Flash memory contention is strange to me: if the flash memory is cached or dual-banked this contention effect should be insignificant. If you really need to take this into account (e.g. for real-time applications), you can calculate worst-case timings for flash access.

`#[shared]` attribute looks extremely helpful for non-uniform memory architectures (even with homogeneous mode). For example, FU540 contains core-local DTIM and ITIMs visible to all the cores, but with non-uniform access times across them. In this case `#[shared]` may indicate that corresponding variable shouldn't be placed in the core-local region. Maybe it's also a good idea to add `#[local(core = X)]` attribute to assign a variable to one of the core-local regions.

LPC55S6x User Manual mentions something called "FMC flash cache", but without any details.

japaric, 2019-07-04:
@Disasm thanks for the info

> if the flash memory is cached or dual-banked this contention effect should be insignificant

I agree.

> However, in real-world devices both I$ and D$ caches are present: K210, FU540 and even FE310.

Is the I$ cache enabled by default in multi-core devices like the K210? Or does it need some setup after a power on reset?

it any case it seems that devices with caches would be OK with merging the input `.text_{i}` and `.text` sections into a single output `.text` provided that an $I cache has been enabled on the whole section.

> LPC55S6x User Manual mentions something called "FMC flash cache", but without any details.

(That chip is rather new and the manual is a bit lacking and has some errors in some parts.)

The ARMv8-M architecture does define caches and registers to perform cache operations but it seems that only the Cortex-M35P cores have built-in caches; the LPC55S6x has 2 Cortex-M33 cores.

Reading the CLIDR (Cache Level ID Register) on an actual device returned all zeros which indicates that no (I or D) caches exist at any of the 7 possible levels.

For the LPC54114 (heterogeneous, M4F + M0+) NXP recommends that one of the cores runs all its code from RAM in one of their application notes as the device has only one Flash bank and no caches. I haven't found a similar recommendation / application note for the LPC55S6x.

> Maybe it's also a good idea to add #[local(core = X)] attribute to assign a variable to one of the core-local regions.

Note that this RFC pertains only to items declared *within* the `#[app]` module and generated by the macro and everything is already core local by default. Letting the user place resources or tasks in core-local memory seems like it would be wrong in most cases (like placing a resource owned by core #0 in memory local to core #1).

Perhaps what you want is something like rust-embedded/cortex-m-rt#164 that can be used in libraries but that's not tied to a particular architecture? (And you can always use `#[linker_section]` to finely control where things go, provided that you are very careful)

japaric, 2019-07-08:
@korken89 @TeXitoi any thoughts on this RFC? This RFC only involves the experimental multi-core API, which we are allowed to change in backwards incompatible ways during the v0.5.x releases, so this may not be what we stabilize at the end but I think the flexibility it allows will let us collect more data while the feature remains experimental (and it's required to get good perf on homogeneous, cache-less devices like the LPC55S69).

korken89, 2019-07-08:
Overall I think this is a great addition!

One question that I am quite sure is not an issue but that I have got stuck on, the `#[shared]` expands to `#[microamp::shared]`, what does this mean?
The thought that comes to my mind is; how will the linker know that the shared variable is the same address for both binaries as the linker puts no guarantees on order of linked variables vs order of variables in the original source files.
I guess there is a trick here I am not seeing to fix the address of shared variables across binaries?

japaric, 2019-07-08:
I think we can FCP (merge) this then.

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi 

---

@korken89 the microamp framework does the heavy lifting. [This blog post](https://blog.japaric.io/microamp/) describes how it works but the TL;DR is that the `#[microamp::shared]` places all variables in a *single* (input) linker section that is forced (using a linker script) to be kept in each compiled binary (`cargo-microamp` compiles the source code several times; one for each different core). Using a single linker section ensures that no variable is GC-ed or reordered by the linker; `cargo-microamp` has a post-build sanity check that inspects all the output ELF files and checks that all the `#[shared]` variables (which live in the output `.shared` linker section) have the same addresses and sizes.

TeXitoi, 2019-07-08:
Even if I'm not aware of these kind of devices, this proposal seemsclean and flexible.

OK for me.

korken89, 2019-07-09:
Thanks for the clarification @japaric !
I'll look more into `cargo-microamp` to get a better understanding, but generating specific linker scripts should indeed to the trick!

japaric, 2019-08-20:
The FCP has passed; this RFC is now in the accepted state. Implementation is in PR #205.

japaric, 2019-09-15:
Done in PR #205

closed 2019-09-15 by japaric