rtic-rs/rtic #213: cortex-m-rtfm-macros does not build on latest nightly 1.38

korken89, 2019-07-11:
The following error is produced, seems to be a simple import change though:

```
   Compiling cortex-m-rtfm-macros v0.4.3

error[E0433]: failed to resolve: unresolved import
   --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/owned-singleton-macros-0.1.0/src/lib.rs:120:46
    |
120 |         let punctuated = Punctuated::<Ident, Token![,]>::parse_terminated(input)?;
    |                                              ^^^^^^^^^
    |                                              |
    |                                              unresolved import
    |                                              help: a similar path exists: `syn::token`
    |
    = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

   Compiling nrf52832-pac v0.8.0
error: aborting due to previous error

For more information about this error, try `rustc --explain E0433`.
error: Could not compile `owned-singleton-macros`.
warning: build failed, waiting for other jobs to finish...
error[E0433]: failed to resolve: maybe a missing `extern crate token;`?
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtfm-macros-0.4.3/src/syntax.rs:32:28
   |
32 |             let _eq_token: Token![=] = input.parse()?;
   |                            ^^^^^^^^^ maybe a missing `extern crate token;`?
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error[E0433]: failed to resolve: maybe a missing `extern crate token;`?
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtfm-macros-0.4.3/src/syntax.rs:59:20
   |
59 |             let _: Token![,] = input.parse()?;
   |                    ^^^^^^^^^ maybe a missing `extern crate token;`?
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error[E0433]: failed to resolve: maybe a missing `extern crate token;`?
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtfm-macros-0.4.3/src/syntax.rs:72:19
   |
72 |     _const_token: Token![const],
   |                   ^^^^^^^^^^^^^ maybe a missing `extern crate token;`?
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error[E0433]: failed to resolve: maybe a missing `extern crate token;`?
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtfm-macros-0.4.3/src/syntax.rs:74:19
   |
74 |     _colon_token: Token![:],
   |                   ^^^^^^^^^ maybe a missing `extern crate token;`?
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error[E0433]: failed to resolve: maybe a missing `extern crate token;`?
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtfm-macros-0.4.3/src/syntax.rs:76:16
   |
76 |     _eq_token: Token![=],
   |                ^^^^^^^^^ maybe a missing `extern crate token;`?
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error[E0433]: failed to resolve: maybe a missing `extern crate token;`?
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtfm-macros-0.4.3/src/syntax.rs:79:18
   |
79 |     _semi_token: Token![;],
   |                  ^^^^^^^^^ maybe a missing `extern crate token;`?
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

[... 208 lines cut ...]
    |
309 |                         gt_token: <Token![>]>::default(),
    |                                    ^^^^^^^^^
    |                                    |
    |                                    unresolved import
    |                                    help: a similar path exists: `syn::token`
    |
    = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error[E0433]: failed to resolve: unresolved import
  --> /home/emifre/.cargo/registry/src/github.com-1ecc6299db9ec823/serde_derive-1.0.94/src/fragment.rs:58:18
   |
58 |                 <Token![,]>::default().to_tokens(out);
   |                  ^^^^^^^^^
   |                  |
   |                  unresolved import
   |                  help: a similar path exists: `syn::token`
   |
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)

error: aborting due to 18 previous errors

For more information about this error, try `rustc --explain E0433`.
error: Could not compile `cortex-m-rtfm-macros`.
```

jonas-schievink, 2019-07-11:
See https://github.com/rust-lang/rust/issues/62562

korken89, 2019-07-11:
Thanks for finding the root cause!

closed 2019-07-11 by korken89