rtic-rs/rtic #214: Reporting of max priority overflow is broken

korken89, 2019-07-11:
Hi, if one creates an app where the assigned priority is above the maximum as (max is 8):

```
#![no_main]
#![no_std]

use nrf52832_hal as hal;
use panic_semihosting as _;
use rtfm::app;

#[app(device = crate::hal::target)]
const APP: () = {
    #[init]
    fn init() {}

    // Fails
    #[interrupt(priority = 9)]
    fn GPIOTE() {}

    /// Also fails
    #[task(priority = 9)]
    fn handle_dw1000_irq() {}
}
```

Gives the following compile error, which is impossible to debug unless one understands the internals of RTFM:
```
    Checking app v0.1.0 (/home/emifre/binary)
error[E0080]: evaluation of constant value failed
  --> src/main.rs:16:1
   |
16 | #[app(device = crate::hal::target)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ attempt to subtract with overflow
```

japaric, 2019-08-20:
This is the documented behavior: https://github.com/japaric/cortex-m-rtfm/blob/fafeeb27270ef24fc3852711c6032f65aa7dbcc0/book/en/src/by-example/resources.md#L72-L77

It *might* be possible to improve the span of the error message but , afaik, the only way to have a custom error message is to have panics in const fn displayed as compile time errors but that's not even possible on nightly and we target the stable channel.

The alternative is to switch to the previous behavior of checking this at runtime instead of at compile time but then you get a panic if the priority is too high.

korken89, 2019-08-20:
Ah, that is unfortunate. But will not a panic in the proc macro become a compile time error? I'm quite sure I used that in smlang.

japaric, 2019-08-21:
> But will not a panic in the proc macro become a compile time error?

That's correct but the *value* of the `#device::NVIC_PRIO_BITS` constant is not available to the proc macro so the check cannot be done in the proc macro itself. What the proc macro does is include in the expansion a (poor man's) compile time check that includes the `#device::NVIC_PRIO_BITS`  *path*.

korken89, 2019-08-21:
Ahh right, thanks for the clarification!

korken89, 2019-08-21:
Having the span point to the right position would be great though if possible

jonas-schievink, 2019-11-09:
The upstream issue for this is https://github.com/rust-lang/rust/issues/43081, right?

korken89, 2019-12-29:
Core issue: Proc macros can evaluate constants. Closing.

closed 2019-12-29 by korken89