rtic-rs/rtic #216: The context/resource syntax is verbose and hard on the eyes

mvirkkunen, 2019-07-17:
In a program that just has `fn main()` and a busy loop I can access resources very concisely like this:

    led_pin.set_high();

In RTFM 0.5.0 the same thing looks like this:

    c.resources.LED_PIN.set_high();

This is pretty verbose and my immediate reaction is that I don't like it. The `c.resources.` prefix gets repetitive really fast, and I don't like that all my resource names are SCREAMING. In practice in my current project I just created a variable for every resource at the start of every function, but this feels like pointless boilerplate code I shouldn't need to write:

    fn some_task(c: some_stack::Context) {
        let mut led_pin = c.resources.LED_PIN;
        // ....
    }

I understand the design decisions for each part of the current syntax are probably as follows:

1. `c`: Magic variables are not immediately obvious. An actual argument is.
2. `resources`: It creates a namespace to avoid conflicts between resources and other things like `spawn`.
3. `LED_PIN`: The SCREAMING is because they're all essentially `static` variables.

All of this *makes sense*, but the end result is just *not nice* - at least in my opinion. Personally I would prefer to just have resources appear magically as lower-case local variables, because I know to avoid name conflicts.

I don't have a perfect or even a good solution to propose but I didn't see any discussion about the increasing verbosity of accessing resources, so I thought I'd open an issue for discussion.

burrbull, 2019-07-17:
Last code is in heterogeneous branch:
https://github.com/japaric/cortex-m-rtfm/blob/heterogeneous/examples/resource.rs

3 item is not actual.

mvirkkunen, 2019-07-17:
@burrbull Nice! I'm already liking this syntax more than the one in the master branch. I didn't really check the other branches besides master before opening this.

burrbull, 2019-07-17:
Maybe if implement `Deref`/`DerefMut` for `Context`:
```
let mut led_pin = (*c).led_pin;
```
I'm not sure.

mvirkkunen, 2019-07-17:
@burrbull If we did that I think the resources could just live directly in the context object without `Deref`, but it would cause the same name conflict issue that I believe the `resources` bit is trying to avoid.

mvirkkunen, 2019-07-17:
Some ideas that may of may not be viable:

The context and resources could be separate function arguments. The macro can parse the function arguments to figure out what arguments each function takes, so you can leave either one (or both!) out if you don't need them. This also gets rid of `_: task::Context` arguments if you really don't need anything.

Are `spawn` and `schedule` really related to the context or are they there just for convenience? Could they be moved elsewhere? This would make the context argument needed even less often and would probably result in many functions only having a "resources" argument.

burrbull, 2019-07-17:
#155

japaric, 2019-09-22:
Hello @mvirkkunen. Sorry for taking so long in answer.

The syntax (in master) has changed since you posted this issue so we would appreciate a new issue in rtfm-rs/rfcs with your feedback on the new syntax.

> The c.resources. prefix gets repetitive really fast

You can always destructure the `Context` type in the task function signature if you wish to avoid the namespacing. For example,

``` rust
#[task(resources = [a, b])] 
fn foo(foo::Context { resources: foo::Resources { a, b }, .. }: foo::Context) {
    a.lock(|_| { b.lock(|_| {}) });
}
```

I'm going to close this issue. We will use the /rfcs repo for discussion and use the issue tracker on this repo for bug reports and tracking progress on implementation work.

closed 2019-09-22 by japaric