rtic-rs/rtic #224: `spawn` inside `lock` section

burrbull, 2019-08-07:
Now when `resources` and `spawn` in one `Context` it's unable to call `spawn` inside `lock` section.
```
    error[E0502]: cannot borrow `*c.resources.adc_cb` as mutable because it is also borrowed as immutable
    ...
    c.spawn.algoritm().unwrap(); 
    - first borrow occurs due to use of `c` in closure
```

korken89, 2019-08-07:
Hi, can you provide a minimal example?
Thanks!

japaric, 2019-08-20:
@burrbull  Move the fields out of the `Context` struct:

``` rust
fn foo(cx: foo::Context) {
    let mut resources = foo.resources;
    let spawn = foo.spawn;

    resources.radio.lock(|radio| {
        spawn.bar();
    });
}
```

Or equivalently destructure the `Context` argument:

``` rust
fn foo(foo::Context { resources, spawn }: foo::Context) {
    resources.radio.lock(|radio| {
        spawn.bar();
    });
}
```

burrbull, 2019-08-20:
Interesting workaround. Thank you.

closed 2019-08-20 by burrbull