rtic-rs/rtic #229: How to specify the order in which interrupts are enabled?

abelykh0, 2019-08-22:
How to specify the order in which interrupts are enabled? 

Use case: 
TIM2: lower priority, WFI in interrupt handler
TIM3: higher priority, runs slightly later than TIM2
Used as a "shock absorber" (to ensure deterministic latency).

For this use case I need to enable TIM3 earlier than TIM2 otherwise everything will idle forever.

perlindgren, 2019-08-23:
Hi

There is currently no means to control the ordering of interrupt enables (nor to manually enable/disable interrupts within the RTFM model).

Wfi in an interrupt handler is not an idiomatic use. (You may WFI in idle, but putting it an an interrupt handler may be dangerous, starving out other low priority tasks.)

I would be interested in your use case, to better understand the need for a shock absorber. What kind of shock are we talking about (bursts of interrupts or...)?

Best regards
  Per

abelykh0, 2019-08-23:
I do agree that it is not a common use case. Can I enter a feature request something in line "Ability to specify the order in which interrupts are enabled"?

For the explanation of the use case, please see http://cliffle.com/blog/glitch-in-the-matrix/.