rtic-rs/rtic #232: Syntax for a static resource that changes type

lthiery, 2019-09-09:
I have a pin that is normally an input, but is repurposed as an output during a certain event. Since input/output are different types, I need to move the pin, use it in its different state, and then put it back.

I am stuck though as an RTFM managed resource is either `mut &` or `&` as far as I can tell. 
[And so my solution looks like this.](https://github.com/lthiery/stm32l0xx-hal/blob/dynamic-pin-type/examples/dynamic_pin.rs).

Is there a nicer way of doing this that I missed?

TeXitoi, 2019-09-10:
You can have the `Option<Pin>` as a ressource, and access it mutably using the lock API.

little-arhat, 2019-09-10:
Similar issue arises when using DMA in rtfm applications. We model this, indeed, using Option<WrapperType>, where wrapper is something like `struct { Busy(..), Ready(..)}`. It's possible then to use Option::take and update resources, like this: https://github.com/copterust/proving-ground/blob/master/dma_int/main.rs#L126.

closed 2019-10-20 by lthiery