rtic-rs/rtic #247: Task / Resource Handles

MabezDev, 2019-09-17:
**The issue**

Coming from an FreeRTOS background, I am used to being able to pass Queue and task handles around such none of the tasks need to know of the other tasks existance. I'd like to be able to do the same thing with RTFM.

**An example scenario:**

A Task offloads a HttpRequest to the HttpClient task by enqueing the request onto a queue that the HttpClient Task is listening to. The client performs the request but how does it pass the result back? The result could be a simple message to say it completed, or it could actually have data to return.

One option is having a queue on the Callee task that the client task can push the result onto, but how do we communicate to the client task which queue to send it back to?

I don't really have any strong thoughts on how to solve this (bar hardcoding values, which I am doing currently). I'm guessing there would be some sort of tagging system such that a tag or handle can resolve to a resource in a safe way.

korken89, 2019-12-29:
Hi, sorry for letting this going missed.
To communicate between tasks you have 3 options:

1. Use `spawn` interface to move data
2. Use a resource to indicate the information
3. Redesign so that both just generates events for a handler task so the handler task has all the information needed to make the distinction

The main issue you are describing is having a runtime selection of resources it seems?
For this I'd probably make a macro which expands to a switch which gives the correct resource based on some condition and panic if none of the conditions fit.

MabezDev, 2020-01-08:
This fell off my radar too, but thanks for getting back to me.

> The main issue you are describing is having a runtime selection of resources it seems?

Pretty much, though I've come to realise this probably isn't feasible with rtfm as inherently it is a runtime check. 

My current solution (which is quite similar to what you suggested) is to tag data with a source, which is just a enum of tasks (currently manually created), something like:

```rust
pub enum TaskList {
    Dns,
    HttpClient,
    // ...
}
```

The transaction of the http client looking up a domain name would go like this:

1. HttpClient creates dns request, tags source as `TaskList::HttpClient`
2. The request is given to the Dns task via any of the three methods you mentioned above, in my case a queue
3. The Dns task completes the request, upon doing so looks up source and runs a callback on the HttpClient resource.

This seems to work pretty nicely. What would be nice is if rtfm could create that task list for me.

PR #436 (2021-03-04): New monotonic
Design document: https://hackmd.io/vWa9GvssR8qBfUYgMZm0CQ

Closes #433 
Closes #432
Closes #427
Closes #426 
Closes #403
Closes #332
Closes #312 
Closes #309 
Closes #299 
Closes #292
Closes #247
Closes #219
closed 2021-03-04 by bors[bot]