rtic-rs/rtic #249: Why are dependencies specified to patch level?

nickray, 2019-09-18:
For instance, `cortex-m-rt = "0.6.9"` is set. But `0.6.10` is the latest release. So someone will have to create an issue, a pull request, a new RTFM release, etc.

If I am not mistaken, setting `cortex-m-rt=0.6` in dependencies would do the right thing, and semver for pre-v1 releases should guarantee that nothing breaks.

Generally, I would expect dependencies to be specified as `0.x` instead of `0.x.y`, what am I missing?

jonas-schievink, 2019-09-18:
Cargo already knows that `0.6.9` is compatible with `0.6.10` and will use the latter. The only way to prevent that is by using a stricter version requirement like `"=0.6.9"`, but rtfm doesn't do that.

nickray, 2019-09-18:
Ah OK, so it's asking for at least `. 9`. Thanks!

closed 2019-09-18 by nickray