rtic-rs/rtic #258: Static Monotonic configuration

joeferner, 2019-10-22:
I have some work in progress code attempting to implement the Monotonic trait for the STM32F0xx.
1. [Modify embedded-hal to include a Monotonic trait](https://github.com/joeferner/embedded-hal/commit/179b4f3a48e82e97d684ae44ca425cdb8d492d78)
2. [Add implementation into stm32f0xx-hal
 for embedded-hal Monotonic trait](https://github.com/joeferner/stm32f0xx-hal/commits/monotonic-timer)
3. [Implement the RTFM Monotonic trait for STM32F0xx](https://github.com/joeferner/cortex-m-rtfm-stm32f0xx)

I created 1 and 2 to provide a means to configure the STM32 timers as Monotonic timers (https://github.com/joeferner/cortex-m-rtfm-stm32f0xx/blob/master/examples/blink.rs#L26).

Now I'm trying to implement `Monotonic.ratio`. Since STM32 timers can run at a variable frequency I would need to update the ratio any time the timer's frequency changes. Since `ratio` is a static method I need to come up with a way to update that static value (https://github.com/joeferner/cortex-m-rtfm-stm32f0xx/blob/master/src/tim.rs#L183).

Maybe you have some thoughts on how this might work.

I'm wondering if it might be better to return a `Monotonic` instance from the RTFM `init` method instead of as an argument to the `rtfm::app` attribute. Then have the monotonic instance available in the `Context` arguments. This would also have the side effect of having better Rust ownership lineage.

japaric, 2019-11-07:
> Since STM32 timers can run at a variable frequency I would need to update the
> ratio any time the timer's frequency changes

Monotonic timers with (runtime) variable frequency are currently unsupported and
are unlikely to be supported any time soon. The main reason is that it's hard to
reason about changes in frequency with the current `Duration` API which uses the
timer tick as the unit time. Some of the questions that arise are: can the
frequency by changed immediately or only when the timer queue is empty (all
previously `schedule`-d tasks have been dispatched)? if the change is not
immediate then calls to `schedule` right after a change in frequency has been
requested are supposed to use the current frequency or the newly requested
frequency? These questions become non issues if the `schedule` API used a non
dilatable / contractable time unit like seconds. @perlindgren has written more
on the topic of wall clock time and "mode changes" (including timers with variable
frequencies) in rtfm-rs/rfcs#23.

---

> means to configure the STM32 timers as Monotonic timers

As per the above reason, applications are expected to use a monotonic timer with
a frequency that remains constant for the whole execution of the program.
That's why the macro uses a stateless Zero Sized Type (the `monotonic` argument)
and a static method (`Montonic::ratio`); they more or less forbid variable
frequency timers.

To deal with configurable timers -- CYCCNT is a *non*-configurable timer; it has
a fixed ratio -- RFC #200 (see the "an example" section) suggests declaring the
monotonic timer "at the top" (in the application, not in a library / HAL). But
there's another way if you really want use a `Monotonic` implementation provided
by a library / HAL (\*). You can declare a *generic* type in the HAL and then
specify a concrete type in the `monotonic` argument. Example below:

``` rust
// crate-name: some-hal

// Zero Sized Type
pub struct Timer<P> {
    // P = clock prescaler
    _marker: PhantomData<P>,_
}

impl<P> rtfm::Monotonic<P>
where
    P: typenum::Unsigned,
{
    fn ratio() -> rtfm::Ratio {
        rtfm::Ratio {
            numerator: 1,
            denominator: P::U32,
        }
    }
}
```

``` rust
// crate-name: some-app

#[rtfm::app(monotonic = some_hal::Timer::<typenum::U2>, /* .. */)]
//                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
//                      concrete (non-generic) type
const APP: () = {
    // ..
};

```

(\*) I would advise against making a HAL depend on `rtfm` just to implement the
`Monotonic` trait because `rtfm` is a rather large dependency that includes a
procedural macro. The idea is to eventually move the `Monotonic` trait into
`rtfm-core` crate which only contains a few traits; I would suggest waiting
until then.

---

> I'm wondering if it might be better to return a Monotonic instance from the
> RTFM init method

It's an option but requires careful consideration. If we turn monotonic into a
value then people are going to expect it to be possible to store state in the
value and use that state in the `Monotonic` methods. The codegen and current
ceiling analysis assumes that the monotonic timer is stateless (perhaps
memory-less is a better word; it's OK to have state in MMIO but not in plain
memory / RAM) so all `Monotonic` methods are run without a surrounding `lock`;
all that would need to be reconsidered if we turn `monotonic` into a value.
Also, it would actually be possible to implement a variable frequency timer,
which we don't support, if we allow stateful monotonic timers

joeferner, 2019-11-07:
Excellent, thanks for the well thought out answer. I'll explore these options and if I come up with something better I'll post my solution.

closed 2019-11-07 by joeferner