rtic-rs/rtic #276: RTFM STM32F3 Discovert Rust

imran82ali, 2019-11-17:
Dear Representative,
We are working on STM32F3 Discovery board, we tried to implement Timer queue using RTFM, after many number of tries we are unable to implement this example on STM32f3 discovery. 
OS Linux Ubuntu 18.04, 
We are using rustc (Version 1.35), 
We are using 
gdb-multiarch, 
arm-none-eabi-gdb 
openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg
thumbv7em-none-eabihf
We have rustup toolchain of stable and nighlty. 
We are stuck in your code. Kindly let us know that your code is compatible with above hardware?
We are following https://rtfm.rs/0.5/book/en/by-example/timer-queue.html.
Thanks,

TeXitoi, 2019-11-17:
It should work.

Please point to a complete repository and explain  clearly what you expect and what you have. Without such information it's difficult to help.

ukashasohail, 2019-11-17:
@TeXitoi repo link https://github.com/rtfm-rs/cortex-m-rtfm/blob/master/examples/schedule.rs

TeXitoi, 2019-11-17:
This file as is can't work on a discovery board. You need at least to give the corresponding PAC.

You also need a Cargo.toml, and some configuration as explained in the RTFM book.

So, to be able to help, I need a full project that I can copy on my computer, and the commands that you are doing, how you plug your board, etc. Without that I can't help you.

tones111, 2019-12-02:
I'm also having trouble running the timer queue example on the STM32F3 discovery board.  It appears that the interrupt handler isn't getting loaded into the vector table.  At the point in time where the task should be getting called the debugger hits a breakpoint within the default handler.

A full example can be found at https://github.com/tones111/rtfm_app/tree/blink
Maybe something is missing from the configuration, but I have yet to find any hints to the cause in the RTFM book.  Thanks in advance for any help.

The output I get in the debugger is...
`enabling cycle counter
init @ Instant(0)
idle @ Instant(1978)
idle @ Instant(5262)
idle @ Instant(8546)
idle @ Instant(11830)
idle @ Instant(15133)
idle @ Instant(18436)
idle @ Instant(21739)
idle @ Instant(25042)
idle @ Instant(28345)`

The debugger then hits a breakpoint at .cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.10/src/lib.rs:555
which is the loop inside DefaultHandler)()

perlindgren, 2019-12-02:
Hi tones11.
I think the problem may be that you need the "rt" feature to load the interrupt vector. 
features = ["stm32f303", "rt"]

I stumbled onto a similar problem, but then the "rt" feature was re-exported through the "hal" implementation.

Let us know how it pans out.

tones111, 2019-12-02:
brilliant. working as expected now.  thanks.

korken89, 2019-12-29:
This is fixed. Closing

closed 2019-12-29 by korken89