rtic-rs/rtic #282: rust-lld: error: no memory region specified for section '.uninit.rtfm7'

abelykh0, 2019-11-20:
Build code from here: https://github.com/abelykh0/stm32f103-vga-rs. Result:

> rust-lld: error: no memory region specified for section '.uninit.rtfm7'

If you comment line 132 in main.rs, the error goes away.

Can you please explain how to solve it? Thanks! I am trying to migrate to 0.5.1 from 0.4.x.

korken89, 2019-11-26:
Hi,

The issue is in your custom linker script which is not compatible with `cortex-m-rt`.
`cortex-m-rt` defined the `.uninit` section which we use, you can find it here: https://github.com/rust-embedded/cortex-m-rt/blob/7c8778f678d444115d74fd4185f32671b67effdc/link.x.in#L132

abelykh0, 2019-11-26:
Thanks a lot! It builds now.

closed 2019-11-26 by abelykh0