rtic-rs/rtic #286: Constructing generic resources (an array of pins)

tshakah, 2019-12-26:
I've used Rust for several years, but mostly for occasional side projects, so I'm not really that good with the language.

I'd like to create an array of pins without specifying explicit types, and am now stuck as the constructed array's contents don't live past the `init` function - I have created a (non-compiling) minimal example that I hope demonstrates what I am trying to do: https://github.com/tshakah/rtfm_stm32_generics

Any help would be appreciated

burrbull, 2019-12-26:
You can make pins generic with [downgrade](https://docs.rs/stm32f1xx-hal/0.5.0/stm32f1xx_hal/gpio/gpioa/struct.PA0.html#method.downgrade) and then collect them into array.

tshakah, 2019-12-26:
Thanks! I had seen that and tried it, but your comment made me look again. I've realised that I forgot to change the type annotation on the type, and after doing that it is now compiling. I have updated the example repository for posterity.

closed 2019-12-26 by tshakah