rtic-rs/rtic #287: Task priority not reflected in NVIC

ggajoch, 2019-12-27:
Hi!

Just started with Rust on embedded (coming from C++).

I've tried to setup a small test app which listens on UART and responds with the message to user - to test the tasks etc.
I am using STM32F0-Disco (STM32F051 cpu).

I've created one task for the interrupt handler (higher priority) which spawns lower priority task after receiving a character. However, I've found that the lower priority task is blocking the USART interrupts - making it into overrun error.

Main parts of my code:
```
#[task(priority = 2, capacity = 10)]
fn time_consuming_task(_: time_consuming_task::Context) {
    print("time consuming task");
}
#[task(priority = 4, binds = USART1, spawn = [time_consuming_task])]
fn usart_handler(_cx: usart_handler::Context) {
    match read(stm32f0::stm32f0x1::USART1::ptr()) {
        Ok(_c) => {
            _cx.spawn.time_consuming_task().unwrap();
        }
        Err(_e) => {
            print("error!");
        }
    }
}
```

First thing I've noticed that priority can be set from 1..255 as per error log, however the hardware has only two bits for priority in NVIC.
`error: this literal must be in the range 1...255`

When I run this code (and send 10 bytes), on the serial port I get:
[image]

Which means that the USART interrupt couldn't preempt the `time_consuming_task`.
Indeed, looking at the NVIC registers we see that those interrupt have the same priority set by the framework:
[image]

However, when I add different task to my program (empty one, never executed) - commented lines 40-42 in `main.rs`:
```
//    #[task(priority = 3)]
//    fn printx2(_: printx2::Context) {
//    }
```

The priorities are set fine and the app works as expected:
[image]
[image]

Full code used here is in the repo: https://github.com/ggajoch/rust-stm32-fun/tree/issue_1.2
I've build it using nightly and stable rust, no difference. Debug/release - the same.

Any clue? I'm more than happy to help but a newbie in Rust :)

jonas-schievink, 2019-12-27:
Is the PAC defining the right number of NVIC bits? ST's SVDs are very bad, they often lie about this.

perlindgren, 2019-12-27:
The current implementation uses BASEPRI to set the "system ceiling" (according to SRP). The M0/M0+ does not provide BASEPRI in HW, so when claiming a shared resource (between tasks of different priority) you will run in a global critical section. In general it should be fine as long as you use lock-free data structures. I haven't looked at the print implementation, maybe there is some locking going on there. 

On the general problem at hand.
One approach to address locking of shared resources on the M0/M0+ is to use the interrupt enable mask (to implement the system ceiling). The company Grepit AB, has implemented this approach, with the plan to open source. Their solution is not complete however, as a subset of the exceptions (like svcall/systick, etc.) have programmable priority (and can be masked). This has yet to be implemented (although, their solution takes care of the "common" case, and likely solve the problem you are facing).

I also did not check your example in detail, the actual priority bit assignments are based on the NVIC priority bits in the SVD file, if they are wrong it will set the wrong bits in the NVIC, and preemption won't be handled correctly.

korken89, 2019-12-27:
Hi, I did some digging.

The reference manual says: `4 programmable priority levels (2 bits of interrupt priority are used)`
However the SVD (`stm32f0x1.svd`) in `stm32-rs` has an error: `<nvicPrioBits>3</nvicPrioBits>`

I have reported this to the `stm32-rs` as a bug: https://github.com/stm32-rs/stm32-rs/issues/324

korken89, 2019-12-27:
I made a PR to `stm32-rs` to fix this issue: https://github.com/stm32-rs/stm32-rs/pull/325

ggajoch, 2020-01-01:
Thanks, I've checked and this fixes the issue 👍

closed 2020-01-01 by ggajoch