rtic-rs/rtic #290: Scheduling a task results in an error, but works anyway

methyl, 2019-12-31:
I'm having a problem with successful task scheduling returning an error:
```
#![no_main]
#![no_std]

use panic_semihosting as _;

use cortex_m_semihosting::hprintln;
use rtfm::cyccnt::{Instant, U32Ext as _};

#[rtfm::app(device = stm32f1xx_hal::stm32, monotonic = rtfm::cyccnt::CYCCNT)]
const APP: () = {
    #[init(spawn = [foo, bar])]
    fn init(mut cx: init::Context) {
        let now = cx.start; // the start time of the system

        hprintln!("init @ {:?}", now).unwrap();

        // Schedule `foo` to run 8e6 cycles (clock cycles) in the future
        cx.spawn.foo().unwrap();

        // Schedule `bar` to run 4e6 cycles in the future
        cx.spawn.bar().unwrap();
    }

    #[task(schedule = [bar])]
    fn foo(cx: foo::Context) {
        hprintln!("foo  @ {:?}", Instant::now()).unwrap();
        cx.schedule
            .bar(Instant::now() + 4_000_000.cycles())
            .unwrap();
    }

    #[task(schedule = [foo])]
    fn bar(cx: bar::Context) {
        hprintln!("bar  @ {:?}", Instant::now()).unwrap();
        cx.schedule.foo(Instant::now() + 8_000_000.cycles());
    }

    extern "C" {
        fn TIM2();
    }
};
```

It results in error when unwrapping `cx.schedule.bar(Instant::now() + 4_000_000.cycles()).unwrap();`:
```
init @ Instant(0)
foo  @ Instant(136)
panicked at 'called `Result::unwrap()` on an `Err` value: ()', src/libcore/result.rs:1189:5
```

When I remove the `unwrap()` call, everything seems to work:
```
init @ Instant(0)
foo  @ Instant(135)
bar  @ Instant(1179)
foo  @ Instant(8002512)
bar  @ Instant(12003890)
```

perlindgren, 2019-12-31:
I would assume that `hprintln!` is too slow. Try a USART or ITM tracing (or just blink a led).

(semihosting is a disaster for any real-time logging and should be avoid besides perhaps debugging panics.)

methyl, 2019-12-31:
It's behaving the same when blinking a LED. Could you help me to understand why performance here may matter?

perlindgren, 2019-12-31:
What I mean is that `hprintln!` may take a lot of CPU cycles, so its not suitable to use in a real-time context. But looking closer at your application, it seems not to be the problem. There could be a capacity issue here though, you have already queued (spawned) task bar, at the point where you schedule bar (from foo). It may be that the implementation of the message passing uses a shared queue for the payload, (I did not look closely at the queue implementation). 

You can just remove the spawn bar from init or set the capacity to 2 for bar.

methyl, 2020-01-01:
The problem persist even if there is a single task that schedules itself, so it's probably not about capacity

methyl, 2020-01-01:
@perlindgren actually, you were right. Increasing capacity worked fine, same for not spawning `bar` in init. I couldn't figure it out because I was missing `cx.core.DWT.enable_cycle_counter();` which was called at some point and it seems it's persistent enough I was unable to track I had it at some point in the program and removed it (since it was working fine without, well).

closed 2020-01-01 by methyl