rtic-rs/rtic #292: Populate an enum with the task names

MabezDev, 2020-01-08:
Mainly for use with task callbacks, see: https://github.com/rtfm-rs/cortex-m-rtfm/issues/247#issuecomment-571972834

AfoHT, 2020-10-15:
In #371 I've included code which generates an enum like the following for all task names:

```
enum Tasks {
    foo,
    bar,
}
```

MabezDev, 2021-03-24:
Hi @AfoHT, I notice that although your PR was merged, it seems the task list code is no longer in master/0.6-alpha?  Any reason for the removal?

AfoHT, 2021-03-25:
Hi!

This seems to have disappeared during refactoring, most likely since there were no tests/examples requiring the task-list.

I'll look into reintroducing it shortly :+1: 

It would be interesting to hear about how these task callbacks work, maybe you could provide some example?

MabezDev, 2021-03-25:
> It would be interesting to hear about how these task callbacks work, maybe you could provide some example?

To be honest I am still looking for a better way of doing this, but essentially I am using it for inter-task communication. This task list simplifies the manual work I have to do.

Take 3 tasks, Sntp client, Http client and Dns Client. Both the Sntp & Http client might need the Dns client to resolve some hostnames before proceeding, the thing is:

* How does the DnsClient report that the Dns Look up has completed
* How does the DnsClient know where the request from?

I am currently solving this via a manually curated `TaskList` enum, and adding a `source: TaskList` parameter which can be attached to the request. E.g Http client would set `source: TaskList::HttpClient`. Then I have a callback trait:

```rust
pub trait DnsCallback {
    fn dns_result(&mut self, res: Result<Ipv4Address, DnsStatus>);
}
```
which I implement on all the client structs, and call the appropriate ones based on the task lisk:

```rust
// match the source task and give back the result
match item.1 {
    TaskList::SntpClient => {
        sntp_client.lock(|sntp_client| sntp_client.dns_result(res))
    }
    TaskList::HttpClient => http_client.dns_result(res),
    _ => unreachable!(),
}
```
Hope that clears up my use case a bit, I welcome more questions or suggestions.

In other RTOS's I've used you can get task/queue handles which can be passed around to tell where to put or take things from. I don't think this model fits in RTIC though as the resource proxies can only last the task duration? I.e they can't be moved out of `taskX::Resources`.

AfoHT, 2021-11-16:
Closing this with reason found in https://github.com/rtic-rs/rtic-syntax/issues/54

PR #436 (2021-03-04): New monotonic
Design document: https://hackmd.io/vWa9GvssR8qBfUYgMZm0CQ

Closes #433 
Closes #432
Closes #427
Closes #426 
Closes #403
Closes #332
Closes #312 
Closes #309 
Closes #299 
Closes #292
Closes #247
Closes #219
closed 2021-11-16 by AfoHT