rtic-rs/rtic #296: Working blinky example for stm32f103 bluepill?

arrowcircle, 2020-01-22:
Hi! 
It looks like there are a lot of changes in RTFM and rust embedded crates.
I am trying to build simple blinky example stm32f103 blue pill, but it does not work.

Also, there are a lot of different examples in the internet, most of them outdated, and almost all of them do not build.

And examples in this repos are very abstract and does not have quickstart code to start using RTFM.

Latest code I found from 2019 is from the users forum, but it does not build.

```rust
#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate panic_semihosting;
use rtfm::{app};
use stm32f1xx_hal::{gpio};
const PERIOD: u32 = 8_000_000;

type LED = gpio::gpioc::PC13<gpio::Output<gpio::PushPull>>;
type BUZZER = gpio::gpioc::PC15<gpio::Output<gpio::PushPull>>;

#[app(device = stm32f1xx_hal::stm32)]
const APP: () = {
    static mut BUZZER: BUZZER = ();
    static mut LED: LED = ();

    #[init(schedule = [foo])]
    fn init() -> init::LateResources {
        schedule.foo(Instant::now() + PERIOD.cycles()).unwrap();
        let device: stm32f1xx_hal::stm32::Peripherals = device;
        let mut flash = device.FLASH.constrain();
        let mut rcc = device.RCC.constrain();
        let mut _afio = device.AFIO.constrain(&mut rcc.apb2);
        let _clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(72.mhz())
            .pclk1(36.mhz())
            .freeze(&mut flash.acr);
        let mut gpioc = device.GPIOC.split(&mut rcc.apb2);
        let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);
        led.set_low();
        let mut buzzer_output = gpioc.pc15.into_push_pull_output(&mut gpioc.crh);
        buzzer_output.set_low();
        // call to the scheduller
        init::LateResources {
            LED: led,
            BUZZER: buzzer_output,
        }
    }

    #[task(schedule = [foo], resources = [BUZZER, LED])]
    fn foo() {

        if resources.LED.is_set_high() {
            resources.LED.set_low();
        } else {
            resources.LED.set_high();
        }

        schedule.foo(scheduled + PERIOD.cycles()).unwrap();
    }

    extern "C" {
        fn EXTI0();
    }
};
```

```txt
cargo build
   Compiling rust-level v0.1.0 (F:\rust-level)
error: this item must live outside the `#[app]` module
  --> src\main.rs:16:5
   |
16 |     static mut BUZZER: BUZZER = ();
   |     ^^^^^^

error: aborting due to previous error

error: could not compile `rust-level`.

To learn more, run the command again with --verbose.
```

part of the cargo.toml
```
[dependencies]
embedded-hal = "0.2.3"
nb = "0.1.2"
cortex-m = "0.6.0"
cortex-m-rt = "0.6.10"
cortex-m-rtfm = "0.5.1"
cortex-m-semihosting = "0.3.3"
panic-halt = "0.2.0"
panic-semihosting = "0.5.3"

# Uncomment for the panic example.
# panic-itm = "0.4.1"

# Uncomment for the allocator example.
# alloc-cortex-m = "0.3.5"

# Uncomment for the device example.
# Update `memory.x`, set target to `thumbv7em-none-eabihf` in `.cargo/config`,
# and then use `cargo build --examples device` to build it.
[dependencies.stm32f1xx-hal]
version = "0.5.3"
features = ["rt", "stm32f103", "medium"]
# this lets you use `cargo fix`!
```
Could please anybody help building basic blinky example with RTFM?

arrowcircle, 2020-01-22:
I found how to make it work using timer. But what is the proper way to use schedule?

```rust
#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate panic_semihosting;
use rtfm;
use cortex_m_semihosting::{hprintln};
use stm32f1xx_hal::prelude::*;
use stm32f1xx_hal::{gpio::{ gpioc::PC13, State, Output, PushPull }, pac, timer::{ Timer, CountDownTimer, Event }};
use embedded_hal::digital::v2::OutputPin;

#[rtfm::app(device = stm32f1xx_hal::stm32, peripherals = true)]
const APP: () = {
    struct Resources {
        led: PC13<Output<PushPull>>,
        timer_handler: CountDownTimer<pac::TIM1>,

        #[init(false)]
        led_state: bool
    }  

    // schedule = [foo]

    #[init()]
    fn init(cx: init::Context) -> init::LateResources {
        hprintln!("init start").unwrap();
        let _device: stm32f1xx_hal::stm32::Peripherals = cx.device;

        let mut flash = _device.FLASH.constrain();
        let mut rcc = _device.RCC.constrain();

        let _clocks = rcc.cfgr.freeze(&mut flash.acr);

        let mut gpioc = _device.GPIOC.split(&mut rcc.apb2);
        let mut led = gpioc.pc13.into_push_pull_output_with_state(&mut gpioc.crh, State::Low);
        led.set_low().unwrap();
        hprintln!("init before schedule").unwrap();
        hprintln!("init after schedule").unwrap();

        let mut timer = Timer::tim1(_device.TIM1, &_clocks, &mut rcc.apb2)
            .start_count_down(1.hz());
        timer.listen(Event::Update);

        init::LateResources { led: led, timer_handler: timer }
    }

    #[task(binds = TIM1_UP, resources = [led, timer_handler, led_state])]
    fn foo(cx: foo::Context) {

        if *cx.resources.led_state {
            cx.resources.led.set_high().unwrap();
            hprintln!("setting high").unwrap();
            *cx.resources.led_state = false;
        } else {
            cx.resources.led.set_low().unwrap();
            hprintln!("setting low").unwrap();
            *cx.resources.led_state = true;
        }
        cx.resources.timer_handler.clear_update_interrupt_flag();
        // cx.schedule.foo(cx.scheduled + PERIOD.cycles()).unwrap();
    }

    #[idle]
    fn idle(_cx: idle::Context) -> ! {
        loop {
            cortex_m::asm::wfi();
        }
    }

    #[interrupt]
    extern "C" {
        fn TIM1();
    }
};
```

perlindgren, 2020-01-22:
In the book `https://rtfm.rs/0.5/book/en/by-example/timer-queue.html` we cover the schedule API. In the examples of the RTFM repo `https://github.com/rtfm-rs/cortex-m-rtfm/tree/master/examples` you find more examples. Look at `schedule.rs` as a starting point.

arrowcircle, 2020-01-23:
So after reading schedule.rs and some encoder examples I "found" working code:

```rust
#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate panic_semihosting;
use rtfm;
use stm32f1xx_hal::prelude::*;
use stm32f1xx_hal::{gpio::{ gpioc::PC13, State, Output, PushPull }};

use embedded_hal::digital::v2::OutputPin;
const PERIOD: u32 = 100_000_000;

use rtfm::cyccnt::{U32Ext as _};

#[rtfm::app(device = stm32f1xx_hal::pac, peripherals = true, monotonic = rtfm::cyccnt::CYCCNT)]
const APP: () = {
    struct Resources {
        led: PC13<Output<PushPull>>,

        #[init(false)]
        led_state: bool
    }  

    #[init(schedule = [foo])]
    fn init(cx: init::Context) -> init::LateResources {
        let mut _cp = cortex_m::Peripherals::take().unwrap();
        let _device: stm32f1xx_hal::stm32::Peripherals = cx.device;

        let mut flash = _device.FLASH.constrain();
        let mut rcc = _device.RCC.constrain();

        let mut _afio = _device.AFIO.constrain(&mut rcc.apb2);
        let _clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(72.mhz())
            .pclk1(36.mhz())
            .freeze(&mut flash.acr);

        _cp.DWT.enable_cycle_counter();
        let now = cx.start;
        cx.schedule.foo(now + PERIOD.cycles()).unwrap();

        let mut gpioc = _device.GPIOC.split(&mut rcc.apb2);
        let mut led = gpioc.pc13.into_push_pull_output_with_state(&mut gpioc.crh, State::Low);
        led.set_low().unwrap();

        init::LateResources { led: led }
    }

    #[task(resources = [led, led_state], schedule = [foo])]
    fn foo(cx: foo::Context) {

        if *cx.resources.led_state {
            cx.resources.led.set_high().unwrap();
            *cx.resources.led_state = false;
        } else {
            cx.resources.led.set_low().unwrap();
            *cx.resources.led_state = true;
        }
        cx.schedule.foo(cx.scheduled + PERIOD.cycles()).unwrap();
    }

    #[idle]
    fn idle(_cx: idle::Context) -> ! {
        loop {
            cortex_m::asm::wfi();
        }
    }
};
``` 

This code works without interrupt and `extern C` part:

```rust
#[interrupt]
extern "C" {
    fn EXTI0();
}
```

Is it ok? What should be corrected in this example?

tshakah, 2020-01-25:
A (almost) bare minimum example in (from what I can tell is) idiomatic RTFM style would be like this:

```rust
#![deny(unsafe_code)]
#![no_std]
#![no_main]

extern crate panic_semihosting;

use embedded_hal::digital::v2::OutputPin;
use rtfm::cyccnt::U32Ext as _;
use stm32f1xx_hal::{gpio, prelude::*};

const PERIOD: u32 = 8_000_000;

#[rtfm::app(device = stm32f1xx_hal::device, peripherals = true, monotonic = rtfm::cyccnt::CYCCNT)]
const APP: () = {
    struct Resources {
        pin: gpio::Pxx<gpio::Output<gpio::PushPull>>,
        led_on: bool,
    }

    #[init(schedule = [toggle])]
    fn init(mut c: init::Context) -> init::LateResources {
        // Initialize (enable) the monotonic timer (CYCCNT) and schedule `toggle`
        c.core.DWT.enable_cycle_counter();
        c.schedule.toggle(c.start + PERIOD.cycles()).unwrap();

        // Get access to the device specific peripherals from the peripheral access crate
        let mut rcc = c.device.RCC.constrain();

        // Acquire the GPIO peripherals
        let mut gpioc = c.device.GPIOC.split(&mut rcc.apb2);

        // Initialise the resources and return
        init::LateResources {
            pin: gpioc.pc13.into_push_pull_output(&mut gpioc.crh).downgrade(),
            led_on: false,
        }
    }

    #[task(schedule = [toggle], resources = [pin, led_on])]
    fn toggle(c: toggle::Context) {
        if *c.resources.led_on {
            // If the pin is on, turn it off
            c.resources.pin.set_low().unwrap();
            *c.resources.led_on = false;
        } else {
            // Otherwise turn it on
            c.resources.pin.set_high().unwrap();
            *c.resources.led_on = true;
        }

        c.schedule.toggle(c.scheduled + PERIOD.cycles()).unwrap();
    }

    extern "C" {
        fn USART1();
    }
};
```

arrowcircle, 2020-01-26:
@tshakah Thanks for the answer. We already figured out demo app and submitter pull request to examples repo https://github.com/rtfm-rs/examples/pull/3

closed 2020-01-26 by arrowcircle