rtic-rs/rtic #299: Adding priority to scheduled task causes problem with resource methods not found

tstellanova, 2020-02-03:
Simply adding a `priority=2` to a scheduled task causes the rtfm expansion to produce code that will not build.  

I have a repeating scheduled task `imu_reader` that processes a series of incoming messages by calling a method `handle_all_messages`.  If I change the task defintion from:
```
 #[task(resources = [i2c1_driver], schedule = [imu_reader])]
```
to 

```
#[task(resources = [i2c1_driver], schedule = [imu_reader], priority = 2)]
```

I get the `cargo build` error:

```
error[E0599]: no method named `handle_all_messages` found for type `resources::i2c1_driver<'_>` in the current scope
   --> src/main.rs:140:34
    |
39  | #[app(device = stm32h7xx_hal::pac,  peripherals = true, monotonic = rtfm::cyccnt::CYCCNT)]
    | ------------------------------------------------------------------------------------------ method `handle_all_messages` not found for this
...
140 |         cx.resources.i2c1_driver.handle_all_messages();
    |                                  ^^^^^^^^^^^^^^^^^^^ method not found in `resources::i2c1_driver<'_>`
```

I may have misunderstood how to configure the rtfm task, but this is unexpected behavior. 
I've attached the original main.rs file as well as the good and bad expansions. 

[rtfm-priority-problem.zip](https://github.com/rtfm-rs/cortex-m-rtfm/files/4149275/rtfm-priority-problem.zip)

perlindgren, 2020-02-03:
Hi
Could it be that its no longer an owned resource, so that you need to lock it before using it. The assymmetric API has both pro:s and con:s. Pro is that it makes it easy to use when you get an owned resource, cons is that an unrelated change to priority breaks the code. (There is an rfc/issue for making the API fully symmetric.)

tstellanova, 2020-02-12:
I admit I don't fully understand the relationship between this simple test app and "the asymmetric API".  For what it's worth, this particular MCU only has one core. In this instance I simply added a priority to an existing task (and did not previously have any priority labels anywhere in the code).  I tried experimenting with locking according to the best available documentation, but with no progress. I subsequently removed the priority label.

perlindgren, 2020-02-12:
The way you access a resource is asymmetric, depending wether there is any task with a higher priority accessing the same resource or not. See the example:

https://github.com/rtfm-rs/cortex-m-rtfm/blob/master/examples/lock.rs

and the book:

/rtfm.rs/0.5/book/en/by-example/resources.html

PTaylor-FluenTech, 2020-04-15:
This threw me at first as well. It also makes things more difficult (for me as a new Rustacean) when passing access of the resource to other functions. Particularly, if I want to pass a Mutex so the lock can be made for a shorter duration only when needed within the other function.

Would it make sense to provide the resource wrapped in an Exclusive when a Mutex is not required? I think that would make the interfaces symmetric.

PR #436 (2021-03-04): New monotonic
Design document: https://hackmd.io/vWa9GvssR8qBfUYgMZm0CQ

Closes #433 
Closes #432
Closes #427
Closes #426 
Closes #403
Closes #332
Closes #312 
Closes #309 
Closes #299 
Closes #292
Closes #247
Closes #219
closed 2021-03-04 by bors[bot]