rtic-rs/rtic #300: scheduling a repeated task ceases after some number of reschedules

tstellanova, 2020-02-12:
I have an [rtfm-based app](https://github.com/tstellanova/stm32h743-rtfm) 
that creates a repeated scheduled task to blink an LED and another repeated task to read an i2c driver.  

This app appears to run the repeated tasks many thousands of times (2-3 minutes real clock time) before the scheduler appears to...stop scheduling the repeated tasks.  At this point the `idle` task runs (verified by setting a breakpoint) and the scheduled tasks never run again.  As far as I can tell, scheduling the repeated tasks does not fail.  

I'm probably doing something obviously wrong here, but this behavior is consistent across stm32h743 and a [similar app running on stm32f401CxUx](https://github.com/tstellanova/test_bno080).

tstellanova, 2020-02-14:
I attempted to boil down the issue to an even simpler reproducible case, but since that hasn't worked, I assume this is a problem with the driver.  I'll close this issue and reopen if I encounter a simpler reproducible case.

closed 2020-02-14 by tstellanova

perlindgren, 2020-02-14:
Sounds like there might be some wrap around problem in play here. If you can come up with a minimal example that be super helpful.