rtic-rs/rtic #311: naming the `#[idle]` function "main" produces a compiler error

japaric, 2020-05-20:
## Steps to reproduce:

Modify the `examples/idle.rs` in this repository like this:

``` diff
     }

     #[idle]
-    fn idle(_: idle::Context) -> ! {
+    fn main(_: main::Context) -> ! {
         static mut X: u32 = 0;

         // Safe access to local `static mut` variable
```

Now the example fails to build:

``` console
$ cargo b --example idle --target thumbv7m-none-eabi
error[E0061]: this function takes 0 arguments but 2 arguments were supplied
  --> examples/idle.rs:19:8
   |
11 | #[rtfm::app(device = lm3s6965)]
   | -------------------------------
   | |
   | supplied 2 arguments
   | defined here
...
19 |     fn main(_: main::Context) -> ! {
   |        ^^^^ expected 0 arguments

error: aborting due to previous error
```

## The problem

If you look into the `rtfm-expansion.rs` file you'll see that the proc-macro produces a `#[no_mangle]` function called "main"; this name collides with the user provided one.

``` rust
fn main(main::Locals { X, .. }: main::Locals, _: main::Context) -> ! {
    // user code
}

const APP: () = {
    use lm3s6965 as _;
    #[no_mangle]
    unsafe extern "C" fn main() -> ! {
        // ..
        main(
            main::Locals::new(),
            main::Context::new(&rtfm::export::Priority::new(0)),
        )
    }
};
```

## The fix

Modify the `main` function inside `const APP` to have a random identifier (it won't be called by user code or proc-macro generated code) and use `export_name` instead of `no_mangle`. So, like this:

``` rust
const APP: () = {
    use lm3s6965 as _;
    #[export_name = "main"]
    unsafe extern "C" fn __rtfm_0() -> ! {
        // ..
        main(
            main::Locals::new(),
            main::Context::new(&rtfm::export::Priority::new(0)),
        )
    }
};
```

Or change the inner `main` function call to use an absolute path: `crate::main(/* args */)`. That may also work.

japaric, 2020-05-20:
The reason one may want to name the `#[idle]` function "main" is better integration with VS Code + Rust-Analyzer. RA gives you this "Run" button (which is sugar for `cargo run --example foo`) when it says a function named "main".

[image]

perlindgren, 2020-05-20:
With the new `mod`  scoping, maybe this will be solved by itself?

japaric, 2020-05-20:
Possibly. I would still like to see this fixed in the v0.5.x series, which use `const APP`, though.

perlindgren, 2020-05-20:
Right, this a bug. Should be fixed in 5.2. (And 6.0)

commit 0ad31107 (2020-06-03): allow handlers to be named 'main'

`#[init]`, `#[idle]` and `#[task]` handlers can now be named `main`

fixes #311
commit 9fe2f4cb (2020-06-03): Merge #315

315: allow handlers to be named 'main' r=korken89 a=japaric

`#[init]`, `#[idle]` and `#[task]` handlers can now be named `main`

fixes #311
commit 4795c1db (2020-06-11): Merge #315

315: allow handlers to be named 'main' r=korken89 a=japaric

`#[init]`, `#[idle]` and `#[task]` handlers can now be named `main`

fixes #311
PR #315 (2020-06-11): allow handlers to be named 'main'
`#[init]`, `#[idle]` and `#[task]` handlers can now be named `main`

fixes #311
closed 2020-06-11 by bors[bot]