rtic-rs/rtic #331: Schedule software tasks and the nRF52

bugadani, 2020-06-17:
Hi!

So I've been scratching my head a lot lately about this. My device was working while using a debugger, but not without. Turns out, my issue is that when the MCU is in WFE/WFI sleep, systick basically stops. This makes periodic tasks at least on the nRF52 completely unusable.

The solution would be to allow changing the timer behind the timer queue to an RTC.

PTaylor-FluenTech, 2020-06-17:
I'm using RTIC with the nRF52 and it works ok. There is a register to configure whether the SysTick timer is halted during WFE/WFI.

bugadani, 2020-06-17:
Unfortunately I can't tolerate the current consumption increase of having to enable the 64MHz clock domain all the time.

PTaylor-FluenTech, 2020-06-17:
I've been trying to find a spec for the difference in current consumption in each configuration. Do you know where to find that or have some measurements?

PTaylor-FluenTech, 2020-06-17:
I have a PR (#312) that allows abstraction of clocks. If I could extend the current abstraction to support interrupt generation, I think that could then allow customization of the clock used for scheduling.

bugadani, 2020-06-18:
Well, "can't tolerate" was a bit stronger than necessary. I don't have a strict power budget, nor the ability to measure such low currents at the moment. I'm rolling with data from the datasheets, however accurate they may be right now, but 50uA is around the current idle consumption of my entire board.

As an additional note: using an RTC would mean that both the Systick timer, and the DWT could be turned off. #312 sounds aweome :)

PTaylor-FluenTech, 2020-06-18:
Without having any hard numbers myself, I agree that being able to turn off the HF clock, SysTick, DWT, etc must make a big (relative) difference in energy consumption.

bugadani, 2020-06-18:
I kind of don't believe my multimeter now, but:
 - I've turned off TQ and configured an RTC to give me ticks for my app
 - System ON with only GPIOTE irqs enabled gave me 670uA total board consumption
 - executing tasks_hfclkstop reduced that to 85.

korken89, 2021-03-04:
This should be fixed with the new monotonic

closed 2021-03-04 by korken89