rtic-rs/rtic #332: Merging SysTick and Monotonic clocks?

PTaylor-FluenTech, 2020-06-18:
Currently, the SysTick timer is used to wake the device when the next scheduled task is due (or SysTick overflows). I understand the desire to make RTIC only reliant on cortex-m resources. I believe it used to be the case that a custom _monotonic_ clock could not be used, but rather the CYCCNT was used exclusively. The _monotonic_ clock was abstracted, so perhaps the _scheduling timer_ (currently SysTick) could also be abstracted.

Another facet of this is that many device peripheral timers/rtcs have a compare register that can generate an interrupt. My current thought is that this would allow one timer to be used both for monotonic and scheduling timer functions.

One possible drawback to this idea is that it may be difficult to continue supporting SysTick as a possible scheduling timer due to its very different functionality.

#331 #312

PTaylor-FluenTech, 2020-06-19:
I believe by adding a `feature` for conditional compilation, the default configuration could be to use the SysTick timer while, when enabled, the clock defined by the `monotonic` parameter could be used also as the scheduling timer. This would allow an implementation to achieve lower power operation by using, for example, an on-chip RTC  as both the monotonic clock and scheduling timer through use of compare interrupts.

korken89, 2020-06-30:
The issue is that one needs a free running timer and a timer with match functionality to get RTICs internals for schedule to work. Usually this means using 2 timers, or at least 2 timer channels.

Was this what you had in mind?

PTaylor-us, 2020-06-30:
You could use one free-running timer with at least one compare interrupt channel (is this what you meant by "match functionality"?). The wakeup is scheduled by loading the compare register with the result of a `wrapping_add` of the current timer count and the duration until the next task. Unfortunately, neither SysTick nor CYCCNT fit the bill, so you currently use them both. It would be nice if there was an _option_ to use one peripheral timer (such as an RTC or general-purpose timer) to handle the scheduling. This could allow very-low-power implementations.

korken89, 2020-09-16:
Indeed, this is a good point.

I have added this as a point on the agenda to look into.

Dirbaio, 2020-09-24:
In nRF52 chips the CYCCNT is not low-power because it requires HFCLK to be active. To do low-power you need to use an RTC, which uses LFCLK so it is low-power when the CPU is sleeping.

PR #436 (2021-03-04): New monotonic
Design document: https://hackmd.io/vWa9GvssR8qBfUYgMZm0CQ

Closes #433 
Closes #432
Closes #427
Closes #426 
Closes #403
Closes #332
Closes #312 
Closes #309 
Closes #299 
Closes #292
Closes #247
Closes #219
closed 2021-03-04 by bors[bot]