rtic-rs/rtic #347: Is it possible to have 'static resource proxies in the idle task ?

jothan, 2020-08-04:
I am spawning tasks inside an async executor that needs tasks to be 'static, I would like to be able to use RTIC resource proxies inside my async task.

Right now, resource proxies in the idle task seem to be of the form &'_ mut Mutex<T>, is this correct ? Since the idle task runs only once, would it be possible to have movable (but !Send, !Sync) resource proxies that are 'static ?

I am falling back on cortex_m::Mutex<RefCell<T>> for now, disabling all interrupts seems like overkill.

korken89, 2020-09-01:
Hi, this is correct that all resources come as `&'_ mut rtic::Mutex<_>`.
Could you provide a small example to discuss around? It would ease us helping in this issue a lot :)

jothan, 2020-09-01:
I'm sharing an async timer queue between the idle task and the interrupt routine that services that queue. The async executor requires that tasks have a 'static lifetime, making access to the ressource proxy impossible.

korken89, 2020-09-01:
Would it be possible to lock the resource outside the executor and have the inner static resource as a resume argument?
I'm not that well into what are the external requirements on resuming an executor.

However it is in the pipeline to fully support async await via native immutable resource proxies. I think you should be able to store the mutable proxy in an immutable proxy and give that to the executor.

jothan, 2020-09-01:
I'm using async-task with a very small home-made executor that uses heapless queues. I don't control the resume argument.

This is not even a personal project at this point, it was more of a proof-of-concept of async with Rust on MCUs.

korken89, 2023-04-15:
Async support is now core in RTIC V2, is you feel this needs more investigation feel free to reopen :)

closed 2023-04-15 by korken89