rtic-rs/rtic #348: Capacity > 127 causes compile failure.

perlindgren, 2020-08-18:
``` Rust
//! examples/capacity.rs

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

use cortex_m_semihosting::{debug, hprintln};
use lm3s6965::Interrupt;
use panic_semihosting as _;

#[rtic::app(device = lm3s6965)]
const APP: () = {
    #[init]
    fn init(_: init::Context) {
        rtic::pend(Interrupt::UART0);
    }

    #[task(binds = UART0, spawn = [foo, bar])]
    fn uart0(c: uart0::Context) {
        c.spawn.foo(0).unwrap();
        c.spawn.foo(1).unwrap();
        c.spawn.foo(2).unwrap();
        c.spawn.foo(3).unwrap();

        c.spawn.bar().unwrap();
    }

    #[task(capacity = 128)] // <- here 
    fn foo(_: foo::Context, x: u32) {
        hprintln!("foo({})", x).unwrap();
    }

    #[task]
    fn bar(_: bar::Context) {
        hprintln!("bar").unwrap();

        debug::exit(debug::EXIT_SUCCESS);
    }

    // RTIC requires that unused interrupts are declared in an extern block when
    // using software tasks; these free interrupts will be used to dispatch the
    // software tasks.
    extern "C" {
        fn SSI0();
    }
};
```

gives:

```
cargo run --example capacity --target thumbv7m-none-eabi
   Compiling cortex-m-rtic v0.5.3 (/home/pln/rust/rtic/cortex-m-rtic)
error: custom attribute panicked
  --> examples/capacity.rs:12:1
   |
12 | #[rtic::app(device = lm3s6965)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: message: UNREACHABLE

error: unused imports: `debug`, `hprintln`
 --> examples/capacity.rs:8:28
  |
8 | use cortex_m_semihosting::{debug, hprintln};
  |                            ^^^^^  ^^^^^^^^
  |
note: the lint level is defined here
 --> examples/capacity.rs:4:9
  |
4 | #![deny(warnings)]
  |         ^^^^^^^^
  = note: `#[deny(unused_imports)]` implied by `#[deny(warnings)]`

error: unused import: `lm3s6965::Interrupt`
 --> examples/capacity.rs:9:5
  |
9 | use lm3s6965::Interrupt;
  |     ^^^^^^^^^^^^^^^^^^^

error: aborting due to 3 previous errors

error: could not compile `cortex-m-rtic`.

To learn more, run the command again with --verbose.:
```

Problem related to the queue length for associated to a priority level. The accumulated queue size of all tasks spawned at a certain priority. Likely related to U8 sized queue.

korken89, 2020-09-01:
The cause of this has been identified, but currently a fix has not been decided on.
It comes from an internal queue which has the sum of all other queues in size which overflows (after rounding up to a power of 2).

korken89, 2021-02-15:
This has now happened to a user on 0.6
Time to fix

korken89, 2021-08-19:
Fixed

closed 2021-08-19 by korken89