rtic-rs/rtic #349: method not found in lazy static resource

jacobrosenthal, 2020-08-19:
Trying to put usbd-serial in rtic

cortex-m-rtic = "~0.5.1"

```rust
    struct Resources {
        usb_bus: &'static UsbBusAllocator<UsbBus>,
        usb_serial: SerialPort<'static, UsbBus>,
        usb_device: UsbDevice<'static, UsbBus>,

        neopixel: NeopixelType,
        lis3dh: Lis3dhType,
    }
```

However when I try to use it
```rust
    #[idle(resources = [neopixel, lis3dh, usb_serial])]
    fn main(c: main::Context) -> ! {
        loop {

            c.resources
                .usb_serial
                .write("Magic Starts!\r".as_bytes())
                .ok();
}
```

```bash
error[E0599]: no method named `write` found for struct `resources::usb_serial<'_>` in the current scope
   --> examples/magic_wand_rtic.rs:169:18
    |
74  | #[rtic::app(device = crate::hal::pac, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
    | --------------------------------------------------------------------------------------------
    | |
    | method `write` not found for this
    | doesn't satisfy `_: embedded_hal::blocking::spi::write::Default<_>`
    | doesn't satisfy `_: pygamer::prelude::_embedded_hal_blocking_spi_Write<_>`
...
169 |                 .write("Magic Starts!\r".as_bytes())
    |                  ^^^^^ method not found in `resources::usb_serial<'_>`
    |
    = note: the method `write` exists but the following trait bounds were not satisfied:
            `resources::usb_serial<'_>: embedded_hal::blocking::spi::write::Default<_>`
            which is required by `resources::usb_serial<'_>: pygamer::prelude::_embedded_hal_blocking_spi_Write<_>`
```

And indeed in the expansion, I dont have a `pub usb_serial: &'a mut SerialPort<'static, UsbBus>` like I would expected, but instead I have a   `pub usb_serial: resources::usb_serial<'a>`

```
mod resources {
    use rtic::export::Priority;
    #[allow(non_camel_case_types)]
    pub struct usb_serial<'a> {
        priority: &'a Priority,
    }
    impl<'a> usb_serial<'a> {
        #[inline(always)]
        pub unsafe fn new(priority: &'a Priority) -> Self {
            usb_serial { priority }
        }
        #[inline(always)]
        pub unsafe fn priority(&self) -> &Priority {
            self.priority
        }
    }
}

#[allow(non_snake_case)]
#[doc = "Resources `main` has access to"]
pub struct mainResources<'a> {
    pub neopixel: &'static mut NeopixelType,
    pub lis3dh: &'static mut Lis3dhType,
    pub usb_serial: resources::usb_serial<'a>,
}
```

Am I doing something wrong or rtic expansion bug?

Code
https://github.com/jacobrosenthal/atsamd/blob/5eed94f4bf7137ad3708382583af13162f539d60/boards/pygamer/examples/magic_wand_rtic.rs

Entire expansion
https://gist.github.com/jacobrosenthal/84752b73a67adbe5c71c54ecbb5e9f71

jacobrosenthal, 2020-08-19:
Oh. I see its in a mutex. Why did that resource get transformed and not usb_device?

closed 2020-08-19 by jacobrosenthal

jacobrosenthal, 2020-08-20:
For anyone who comes after me, rtfm :)

>> Where a critical section is required the framework hands out a resource proxy instead of a reference. This resource proxy is a structure that implements the Mutex trait. The only method on this trait, lock, runs its closure argument in a critical section.

Ie if you use a resource in 2 tasks, it transforms to a mutex

AfoHT, 2020-08-24:
Currently the task with the highest priority get the reference while other tasks needs to acquire the mutex-lock.

There's ongoing [work](https://github.com/rtic-rs/cortex-m-rtic/tree/task_local_experiment) streamlining this to allow for a symmetric API where this "surprise" is not as bad.