rtic-rs/rtic #351: 0.5.4 breaks nrf-hal

jonas-schievink, 2020-08-27:
Since 0.5.4 was released, we're seeing this compilation error for the [rtic-demo](https://github.com/nrf-rs/nrf-hal/blob/master/examples/rtic-demo/src/main.rs) in nrf-hal:

```
error[E0433]: failed to resolve: use of undeclared type or module `cortex_m`
  --> examples/rtic-demo/src/main.rs:22:1
   |
22 | #[app(device = crate::hal::pac)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ use of undeclared type or module `cortex_m`
```

To reproduce, check out the [nrf-hal](https://github.com/nrf-rs/nrf-hal) repo and run:

```
rm Cargo.lock
cd examples/rtic-demo
cargo build
```

AfoHT, 2020-08-27:
That is interesting, apart from the test-suite and examples the only change is found in #338 

I found that by adding 

```
cortex-m = "0.6.3"
```
to `[dependencies]` makes it compile happily.

jonas-schievink, 2020-08-27:
I think #338 used the wrong path for the fix, it needs to go through the `export` module https://github.com/rtic-rs/cortex-m-rtic/pull/338#discussion_r478377971

jonas-schievink, 2020-08-28:
Fixed in 0.5.5

closed 2020-08-28 by jonas-schievink