rtic-rs/rtic #354: Entry point and rust-analyzer

bugadani, 2020-08-30:
Right now, RTIC does not integrate well with rust-analyzer's entry point detection, which in turn means, that the handy "Run >" button is missing in VS Code for projects that use RTIC.

It would be really nice if this could be somehow resolved.

korken89, 2020-09-01:
Hi, could you provide more information on what rust-analyzer is expecting or documentation of this feature?
After proc-macro expansion we are using the same as `cortex-m-rt` with our own `main` function, so I'm wonder what more is required.

bugadani, 2020-09-01:
Maybe I should've checked before opening this issue, this is already known: https://github.com/rust-analyzer/rust-analyzer/issues/4771

closed 2020-09-01 by bugadani