rtic-rs/rtic #361: Panic from init (rtic 0.5.5)

HarryMakes, 2020-09-16:
Hi, I am working on a Cortex-M7-based project that has previously been functional before we update cortex-m-rtic from 0.5.3 to 0.5.5. I suspect that some changes have broken our RTIC implementation ([link for reference](https://github.com/quartiq/stabilizer/blob/v0.4.1/src/main.rs#L160)).

### Environment
* CPU: STM32H743ZI
* Cargo: 1.48.0-nightly (875e01232 2020-09-08)
* Rustc: 1.48.0-nightly (6af1bdda5 2020-09-15)

### Error
GDB backtrace:
```
#0  0x08020c90 in rust_begin_unwind (_info=<optimized out>)
#1  0x08017532 in core::panicking::panic_fmt () at library/core/src/panicking.rs:85
#2  0x0801f0d0 in core::option::expect_failed () at library/core/src/option.rs:1213
#3  0x08014674 in stabilizer::init (c=...)
    at /rustc/6af1bdda54abc9e919fc1137411dfc4311e05649/library/core/src/option.rs:370
#4  0x0801741c in stabilizer::APP::main () at src/main.rs:160
```

Using `next` in GDB, the program panics after:
```
...
halted: PC: 0x08000334
537	                *SCB_CPACR | SCB_CPACR_FPU_ENABLE | SCB_CPACR_FPU_USER,
(gdb) n
Info : halted: PC: 0x08000336
halted: PC: 0x08000336
Info : halted: PC: 0x0800033a
halted: PC: 0x0800033a
535	            core::ptr::write_volatile(
(gdb) n
Info : halted: PC: 0x0800033c
halted: PC: 0x0800033c
550	            trampoline()
(gdb) n
Info : halted: PC: 0x08020c88
halted: PC: 0x08020c88
^C
Program received signal SIGINT, Interrupt.
0x08020c96 in rust_begin_unwind (_info=<optimized out>)
```

korken89, 2020-09-16:
Hi, I see the issue.

This line is undefined behavior and was fixed in the latest release: https://github.com/quartiq/stabilizer/blob/5d41fa0aa981c68372464c016e9e7a4b222dfc06/src/main.rs#L206

As you have the `peripherals = true` flag set you can find the peripherals in the context `c.device`.
Is explained more here: https://rtic.rs/0.5/book/en/by-example/app.html#init

Tell me if you experience more problems!

HarryMakes, 2020-09-16:
Thanks, it seems to be working now.

closed 2020-09-16 by HarryMakes