rtic-rs/rtic #362: No compile error when `rt` feature of HAL isn't active

diondokter, 2020-09-22:
Hello all,

I've spent my evening bashing my head against a problem.
I had a task bound to an interrupt, but instead of the interrupt, the default handler was called.

At the same time, all exceptions would work fine.

The problem ended up being that I had forgotten to enable the `rt` feature in the HAL which in turn didn't generate the vector table fully. The exceptions were there, but the interrupts weren't.
The code, however, compiled fine even though it's obviously wrong.

For usability's sake, this could be an error.

I don't know how one could best detect this though...

korken89, 2020-09-23:
Hi, thanks for the issue!

Unfortunately RTIC cannot detect this as it simple interfaces with the HAL/PAC to get the correct interrupt numbers.
If the PAC has not had the `rt` flag enabled it means that you as the user are responsible for providing the interrupt vector and RTIC assumes the user has provided it.

If you have ideas for automatic checking of this it would be interesting!
One could look into providing a default handler in RTIC that simply panics and says that the interrupt is not bound.

AfoHT, 2020-09-23:
The default handler doesn't sound all too bad, and instead of silently failing it would get explicit something is bad.

korken89, 2020-09-23:
We can look into this, lets just be careful with `fmt` bloat.
But a panic which says "Unbound interrupt detected, is the `rt` feature enabled in the PAC/HAL?" should be helpful and optimize away well if the panic handler requests it.

diondokter, 2020-09-23:
I've found something interesting!

The `rt` feature activates the `device` feature in cortex-m-rt.

https://docs.rs/cortex-m-rt/0.6.13/cortex_m_rt/#device 

When the `device` feature is **not** active, then cortex-m-rt creates its own interrupt table that you can find at `cortex_m_rt::__INTERRUPTS`.
But when you do activate the `device` feature, then it will forego creating the interrupt table as that responsibility is left to the user/pac/hal.

So if you want to generate an error, then check if the `cortex_m_rt::__INTERRUPTS` value exists. If it does, then the interrupts have not been set up by the `rt` feature.
It feels like there should be a way to do this at compile time...

https://docs.rs/cortex-m-rt/0.6.13/src/cortex_m_rt/lib.rs.html#709

The only question that remains is whether manual setup of interrupts is a supported usecase for RTIC.

korken89, 2020-09-23:
Unfortunately I am not aware of any way to check for the non-existence of `cortex_m_rt::__INTERRUPTS` at compile time.

korken89, 2020-10-16:
This is now fixed in https://github.com/rtic-rs/cortex-m-rtic/pull/394

closed 2020-10-16 by korken89