rtic-rs/rtic #364: Regression in master after multicore removal

korken89, 2020-09-24:
I have found a regression after the multicore removal.
The following example that I added does not compile any more on `master`: https://github.com/rtic-rs/cortex-m-rtic/blob/schedule_regression/examples/double_schedule.rs

It fails with the error:
```
error[E0574]: expected struct, variant or union type, found static `P1_RQ`
  --> examples/double_schedule.rs:11:1
   |
11 | #[rtic::app(device = lm3s6965, monotonic = rtic::cyccnt::CYCCNT)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not a struct, variant or union type
   |
   = note: this error originates in an attribute macro (in Nightly builds, run with -Z macro-backtrace for more info)
```

And looking at the task dispatcher in the expanded code, one finds that the `static mut P1_RQ` is used as if it was a type:
```rust
    #[no_mangle]
    unsafe fn SysTick() {
        use rtic::Mutex as _;
        #[doc = r" The priority of this handler"]
        const PRIORITY: u8 = 1u8;
        rtic::export::run(PRIORITY, || {
            while let Some((task, index)) = (TQ {
                priority: &rtic::export::Priority::new(PRIORITY),
            })
            .lock(
                #[inline(always)]
                |tq| tq.dequeue(),
            ) {
                match task {
                    T::task1 => {
                        (P1_RQ {
                            priority: &rtic::export::Priority::new(PRIORITY),
                        })
                        .lock(|rq| rq.split().0.enqueue_unchecked((P1_T::task1, index)));
                        rtic::pend(stm32l4xx_hal::stm32::Interrupt::DFSDM1);
                    }
                    T::task2 => {
                        (P1_RQ {
                            priority: &rtic::export::Priority::new(PRIORITY),
                        })
                        .lock(|rq| rq.split().0.enqueue_unchecked((P1_T::task2, index)));
                        rtic::pend(stm32l4xx_hal::stm32::Interrupt::DFSDM1);
                    }
                }
            }
        });
    }
```

Could you have a look at this @AfoHT ?

AfoHT, 2020-09-25:
What a coincidence, been tinkering with this yesterday. Nice you also found this.

Not only the double schedule fails, but all examples using schedule.

I've started to modify the test suite to actually check for these too, since currently the examples requiring the `__v7` feature is never checked.

AfoHT, 2020-09-25:
Fixed in https://github.com/rtic-rs/rtic-syntax/pull/29

closed 2020-09-25 by AfoHT