rtic-rs/rtic #369: Book include formatting is.. weird.

dcarosone, 2020-10-01:
In the book, all the code and console examples follow a pattern like so:
    ``` rust
    {{#include ../../../../examples/resource.rs}}
    ```
    
    ``` console
    $ cargo run --example resource
    {{#include ../../../../ci/expected/resource.run}}```

Note that the console block is terminated with the backticks at the end of the include line, not on a new line.  This breaks the rendering, both for md previews (e.g. in vscode or github's own browsing) and in the output from `mdbook serve`.  All the following md source up to the next set of backticks gets put in as console text.

I don't understand why, but it's consistent and clearly deliberate for some reason. My guess was something related to suppressing an extra newline in the run output, perhaps related to a difference in Windows vs Unix line endings?

This has been this way for a long time. It obviously isn't coming out this way in the process that rendered the pages on rtic.rs. Maybe that's not broken because it was rendered on Windows with different line endings? Maybe it's a change in more recent `mdbook`?

If anyone knows more, please comment. If the reason is still valid, then just close this issue, otherwise I'll make a PR to add in the extra line endings and fix it that way if it's no longer needed, or noone knows why anymore.

AfoHT, 2020-10-01:
Nice find! Sound reasonable this is some strange leftover.

Lets see if latest version of mdBook can handle this odd layout. #370 

Unless anybody knows what reason would require this broken markdown I'm all for fixing it as well.

korken89, 2020-10-01:
Agreed, fixing it would be great!

dcarosone, 2020-10-01:
For whatever it's worth, I have `mdbook 0.4.3` which `cargo install-update` says is current.

closed 2020-10-02 by bors[bot]

dcarosone, 2020-10-10:
This now appears to have shown up on the website rendering, including for old versions of the book.

I'm not sure how they're generated, but I'll guess that the merge of #370 is involved: maybe the CI config there gets used to render each of the older branches at once?

commit baa2edfe (2020-10-02): close console text blocks on a new line

fixes #369
commit 97eae45f (2020-10-02): Merge #375

375: close console text blocks on a new line r=AfoHT a=dcarosone

fixes #369
PR #375 (2020-10-02): close console text blocks on a new line
fixes #369
commit 66362498 (2020-10-12): close console text blocks on a new line

fixes #369