rtic-rs/rtic #376: Code generation issue?

BBScholar, 2020-10-03:
I have an struct called "encoder" within the [Resources](https://github.com/BBScholar/CAN_ENCODER_RUST/blob/bfcbeca6e50d3cc6edec3e8cfe604ccd8948df6f/src/main.rs#L96) struct in my RTIC application. I'm getting the following errors when I try to call methods from my Encoder struct from software tasks.

```
    #[task(resources=[eeprom, encoder])]
    fn mem_tx(cx: mem_tx::Context) {
        // Send data to eeprom memory
        // this will be called on shutdown
        let eeprom = cx.resources.eeprom;
        let mut encoder = cx.resources.encoder;

        // get variables
        let ticks = encoder.ticks();
        let inverted = encoder.inverted();
        let absolute_offset = encoder.absolute_offset();

        // write to eeprom
        eeprom.write_data(memory::Address::Ticks as u8, ticks);
        eeprom.write_bool(memory::Address::Polarity as u8, inverted);
        eeprom.write_data(memory::Address::AbsoluteOffset as u8, absolute_offset);
    }
``` 

```
error[E0599]: no method named `ticks` found for struct `resources::encoder<'_>` in the current scope
   --> src/main.rs:344:29
    |
92  | #[app(device=stm32f1::stm32f103, peripherals = true, monotonic = rtic::cyccnt::CYCCNT)]
    | --------------------------------------------------------------------------------------- method `ticks` not found for this
...
344 |         let ticks = encoder.ticks();
    |                             ^^^^^ method not found in `resources::encoder<'_>`

error[E0308]: mismatched types
   --> src/main.rs:373:62
    |
373 |         let encoder: encoder::Encoder<hardware_types::SPI> = cx.resources.encoder;
    |                      -------------------------------------   ^^^^^^^^^^^^^^^^^^^^ expected struct `Encoder`, found struct `resources::encoder`
    |                      |
    |                      expected due to this
    |
    = note: expected struct `Encoder<Spi<stm32f1xx_hal::pac::SPI1, Spi1NoRemap, (PA5<Alternate<PushPull>>, PA6<Input<Floating>>, PA7<Alternate<PushPull>>), u16>>`
               found struct `resources::encoder<'_>`
```

I checked the output of cargo expand, and found that RTIC has generated the following code. I tried changing the name of my encoder resource to a few other things (enc, quad), but still get the same generated struct with the same name as my resource no matter what I change it to.

```
mod resources {
    use rtic::export::Priority;
    #[allow(non_camel_case_types)]
    pub struct encoder<'a> {
        priority: &'a Priority,
    }
    impl<'a> encoder<'a> {
        #[inline(always)]
        pub unsafe fn new(priority: &'a Priority) -> Self {
            encoder { priority }
        }
        #[inline(always)]
        pub unsafe fn priority(&self) -> &Priority {
            self.priority
        }
    }
}

```

I doubt the implementation of my Encoder struct would be causing this issue but I will link it [here](https://github.com/BBScholar/CAN_ENCODER_RUST/blob/master/src/encoder.rs) anyways. I have similar structs within my resource declaration, but I'm not having the same issue with them. Is there something I'm missing? 

I'm fairly new to rust, so excuse me if this is an error on my part.

korken89, 2020-10-03:
Hi, it looks like you have forgot to lock the resource.
Most likely you have changed priorities in the system so that another task accesses the encoder as well, but with a higher priority than this task.

The fix should look something like this:

```rust
let mut encoder = cx.resources.encoder;

// get variables
let (ticks, inverted, absolute_offset) = encoder.lock(|encoder| {
    (
        encoder.ticks(),
        encoder.inverted(),
        encoder.absolute_offset(),
    )
});
```

BBScholar, 2020-10-03:
Ah. That makes much more sense. Thank you for your help!

closed 2020-10-03 by BBScholar