rtic-rs/rtic #386: RTIC app breaks I2C for STMF7xx

ctron, 2020-10-07:
Using I2C in combination with an RTIC app breaks the communication when using stm32f7xx.

The result is a timeout when doing e.g. an I2C WriteRead operation:

~~~
10:21:15.987 panicked at 'called `Result::unwrap()` on an `Err` value: WriteReadError(WouldBlock)', examples/rtic.rs:163:70
~~~

Where `WouldBlock` is the error that comes out of the (blocking!) I2C `write_read` operation, due to the fact that the timeout expired.

Using the same code, I have the following outcomes:

| | `stm32f411` | `stm32f723` |
|---|---|---|
| Normal | :heavy_check_mark: | :heavy_check_mark:  |
| RTIC | :heavy_check_mark: | :x: |

Code: https://github.com/drogue-iot/drogue-bme680/blob/main/examples/rtic.rs

korken89, 2020-10-08:
Hi, this sounds like an HAL error and not related to RTIC.
From what I can see you are not using interrupts, which is the only part that RTIC takes control over.

Without more information this will be tough to debug.

ctron, 2020-10-09:
> Hi, this sounds like an HAL error and not related to RTIC.
> From what I can see you are not using interrupts, which is the only part that RTIC takes control over.

But it looks like RTIC changes something in the environment that breaks I2C. True, I may break some assumption in the F7xx HAL crate.

> 
> Without more information this will be tough to debug.

Just let me know what information can help and I try to provide it.

korken89, 2020-10-09:
Unfortunately I do not have good input for you. 
The only thing you can do is compare the expanded RTIC code and the I2C driver, see if anything is touching the same core parts.
RTIC provides all parts which are safe to use via `cx.core` and `cx.device` in `#[init]`. Most likely the HAL is using `unsafe` for something that collides.

One last idea, what timeout do you have set? After a quick look at the HAL it's checking at core clock speed so the timeout needs to be significant.

korken89, 2021-03-04:
Issue cleanup: Please reopen if this is still active.

closed 2021-03-04 by korken89