rtic-rs/rtic #387: Unable to conditionally use hardware tasks (binds)

ctron, 2020-10-07:
Assuming you want to create an application for multiple targets, you might want to write the following code:

~~~rust
    #[cfg(feature = "stm32f411")]
    #[task(binds = USART2, priority = 20, resources = [ingress])]
    fn usart2(ctx: usart2::Context) {
        if let Err(b) = ctx.resources.ingress.isr() {
            info!("failed to ingress {}", b as char);
        }
    }
    #[cfg(feature = "stm32f723")]
    #[task(binds = UART5, priority = 20, resources = [ingress])]
    fn uart5(ctx: uart5::Context) {
        if let Err(b) = ctx.resources.ingress.isr() {
            info!("failed to ingress {}", b as char);
        }
    }
~~~

However, it seems that `cfg` is ignored by the macro, and so you get the error message that `UART5` cannot be found in the `stm32f4xx_hal` crate.

korken89, 2020-10-08:
Hi, this is a known limitation of proc-macros in Rust today.
We have rudimentary  `#[cfg]` support, but a lot will not work unfortunately.

I'll ping @AfoHT that has more experience here.

What I usually do for HAL examples is that I select a specific target in the `Cargo.toml` for the RTIC examples (you can see how I did it in the stm32l4xx HAL).

little-arhat, 2021-02-17:
(Linking related issue https://github.com/rtic-rs/cortex-m-rtic/issues/197)

little-arhat, 2021-09-09:
Linked issue was closed, but it's not clear what is the solution. Maybe worth changing docs?

perlindgren, 2021-11-30:
We close this issue for now, see [#197](https://github.com/rtic-rs/cortex-m-rtic/issues/197).

closed 2021-11-30 by perlindgren