rtic-rs/rtic #389: Allow Resources to take type parameters

bradleyharden, 2020-10-09:
This might seem like a strange feature to request. After all, if `Resources` is generic, where do you supply its concrete types? But I hope to convince you that it can, in fact, make sense.

In my case, I have a type `P` that uses 6 type parameters to keep track of its state at compile time. Each concrete type supplied as a type parameter is zero-sized, so the entire `P` struct only exists at compile time. To create instances of `P`, I use the builder pattern, so you can supply the type parameters in pieces, and you only need to supply the ones that differ from the default. Even though it has 6 type parameters, type inference and the builder pattern makes `P` fairly ergonomic.

To encapsulate instances of `P`, I implement a `Sealed` trait `Trait` on all possible versions of `P`. Then, when I want to use an instance of `P`, I don't have to enumerate all 6 type parameters. Instead, I can simply use `Trait` as a bound. On top of that, `Trait` contains an associated type for each type parameter on `P`, so you can always recover the type parameters of `P` if necessary.

Ideally, I would like to use the same design pattern in `Resources`. Basically, I want to be able to do this.

```rust
struct P<A, B, C, D, E, F> {
    // ...
}

trait Trait {
    type A;
    // ...
    type F;
}

impl<A, B, C, D, E, F> Trait for P<A, B, C, D, E, F> {
    type A = A;
    // ...
    type F = F;
}

#[rtic::app()]
const APP: () = {
    struct Resources<Z: Trait> {
        p: Z
    }

    #[init]
    fn init(_: init::Context) -> init::LateResource<impl Trait> {
        let p = // create an instance of P
        init::LateResource{ p }
    }
}
```

With this approach, I don't have to explicitly write out the type parameters in `Resources`. It also eliminates duplication. I can specify the type parameters once, in the code, and `Resources` takes whatever it gets.

Is this something you could support in the future?

AfoHT, 2020-10-12:
Interesting!

In latest master of RTIC the static name `struct Resources` has been changed to become an attribute: `#[resources]`.

Thus the name "Resources" is not important, and the `<Z: Trait>` part would hopefully just be passed along.

Maybe you could give it try and see if works, otherwise we can investigate what is missing.

bradleyharden, 2020-10-12:
Ok, I'll try it out. Is there documentation for `master`? I guess I can probably build it from `master`, but I haven't done that before. Or maybe the documentation isn't even up-to-date? I'll see if I can figure it out.

bradleyharden, 2020-10-12:
Hmm, it doesn't seem to work. As far as I can tell, it completely ignores the type parameter. If I try to use the type parameter within the `Resources`, it tells me it doesn't know what the type parameter is. And adding `<impl Trait>` to the `LateResources` return value has no effect, regardless of whether or not `Resources` takes a type parameter.

AfoHT, 2020-10-13:
> Ok, I'll try it out. Is there documentation for `master`? I guess I can probably build it from `master`, but I haven't done that before. Or maybe the documentation isn't even up-to-date? I'll see if I can figure it out.

The latest book and API docs can be found here: https://rtic.rs/dev/book/en/, but also built locally with `cargo doc`/mdBook (for the book)

AfoHT, 2020-10-15:
> Hmm, it doesn't seem to work. As far as I can tell, it completely ignores the type parameter. If I try to use the type parameter within `Resources`, it tells me it doesn't know what the type parameter is. And adding `<impl Trait>` to the `LateResources` return value has no effect, regardless of whether or not `Resources` takes a type parameter.

Could you link to some example using your proposed setup? I would look in `target/rtic-expansion.rs` (use cargo fmt to get it readable).

korken89, 2020-10-16:
With the move we have made to attributes I do not thing it's impossible to make the resources struct generic.
E.g. so one do not have to give concrete types for HAL stuff where it gets very verbose.

Let's keep this in mind.

bradleyharden, 2020-10-18:
See minimal example [here](https://github.com/bradleyharden/rtic_389_example). It looks like the type parameters don't make it through the macro.

Original:

```rust
struct PadsT<S, I, P0, P1, P2, P3> {
    sercom: PhantomData<S>,
    ioset: PhantomData<I>,
    pad0: PhantomData<P0>,
    pad1: PhantomData<P1>,
    pad2: PhantomData<P2>,
    pad3: PhantomData<P3>,
}

trait Pads {
    type Sercom;
    type IoSet;
    type Pad0;
    type Pad1;
    type Pad2;
    type Pad3;
}

impl<S, I, P0, P1, P2, P3> Pads for PadsT<S, I, P0, P1, P2, P3> {
    type Sercom = S;
    type IoSet = I;
    type Pad0 = P0;
    type Pad1 = P1;
    type Pad2 = P2;
    type Pad3 = P3;
}

#[rtic::app(device = crate::pac)]
mod app {

    #[resources]
    struct Resources<P: Pads> {
        p: P
    }

    #[init]
    fn init(_: init::Context) -> init::LateResources<impl Pads> {
        let p = PadsT {
            sercom: PhantomData::<u8>,
            ioset: PhantomData::<u16>,
            pad0: PhantomData::<u32>,
            pad1: PhantomData::<i8>,
            pad2: PhantomData::<i16>,
            pad3: PhantomData::<i32>,
        };
        init::LateResources{p}
    }

}
```

Expanded:

```rust
#[allow(non_snake_case)]
fn init(_: init::Context) -> init::LateResources {
    let p = PadsT {
        sercom: PhantomData::<u8>,
        ioset: PhantomData::<u16>,
        pad0: PhantomData::<u32>,
        pad1: PhantomData::<i8>,
        pad2: PhantomData::<i16>,
        pad3: PhantomData::<i32>,
    };
    init::LateResources { p }
}
#[doc = r" Resources initialized at runtime"]
#[allow(non_snake_case)]
pub struct initLateResources {
    pub p: P,
}
#[allow(non_snake_case)]
#[doc = "Initialization function"]
pub mod init {
    #[doc(inline)]
    pub use super::initLateResources as LateResources;
    #[doc = r" Execution context"]
    pub struct Context<'a> {
        #[doc = r" Core (Cortex-M) peripherals"]
        pub core: rtic::export::Peripherals,
        #[doc = r" Critical section token for init"]
        pub cs: rtic::export::CriticalSection<'a>,
    }
    impl<'a> Context<'a> {
        #[inline(always)]
        pub unsafe fn new(core: rtic::export::Peripherals) -> Self {
            Context {
                cs: rtic::export::CriticalSection::new(),
                core,
            }
        }
    }
}
#[doc = r" Unused"]
#[doc = r" Implementation details"]
mod app {
    #[allow(non_snake_case)]
    use super::init;
    #[allow(non_snake_case)]
    use super::initLateResources;
    #[doc = r" Always include the device crate which contains the vector table"]
    use crate::pac as you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml;
    #[doc = r" User code from within the module"]
    #[doc = r" User code end"]
    #[doc = r" app module"]
    #[no_mangle]
    unsafe extern "C" fn main() -> ! {
        let _TODO: () = ();
        rtic::export::interrupt::disable();
        let mut core: rtic::export::Peripherals = rtic::export::Peripherals::steal().into();
        core.SCB.scr.modify(|r| r | 1 << 1);
        let late = crate::init(init::Context::new(core.into()));
        rtic::export::interrupt::enable();
        loop {
            rtic::export::wfi()
        }
    }
}
```

bradleyharden, 2020-11-07:
I realize it's probably not a priority, but I wanted to ping this issue one more time. It seems like @AfoHT expected the type parameter to get passed along, but it doesn't look like that's the case. Has anyone had a chance to see why?

AfoHT, 2020-11-08:
I started looking into this but got preempted by work.

It does not currently work since the resources are collected into a struct, nothing is done about the type parameters. I'm not sure how tricky it is to bundle along the parameter, but unfortunately I do not have the time available at the moment. In `src/parse/app.rs:277` the parsing of the resource struct is done.

VersBinarii, 2020-11-27:
I also came across this. It would be great to be able to specify type parameters on the Resource struct.
Its handy when using external embedded_hal  libraries. 

With latest master it fails to compile: "cannot find type '' in this scope".

korken89, 2020-11-30:
I agree, this should be a priority for better iteroperability with the PAC/HALs that are very types based.

korken89, 2020-12-07:
With the new proposed resource syntax (https://github.com/rtic-rs/rfcs/issues/46), this should be much easier to realize.
The issue today is analyzing the resource structure and figure out which type belong to which field in the late vs early resources.

bradleyharden, 2020-12-15:
I just realized that this may not be possible. I don't think you can declare a static using an inferred type or an `impl Trait`. I just tried both of those out on the playground, but neither worked.

```rust
trait Trait {}

struct A;
impl Trait for A {}

struct Struct<T: Trait> {
    field: T,
}

// Neither of these works...
static TEST1: Struct<_> = Struct { field: A};
static TEST2: Struct<impl Trait> = Struct { field: A };
```

If neither of these works, then you need to know the concrete type as part of the expansion, right? And if that's true, then this isn't possible.

Do you see another way to do it?

korken89, 2021-03-04:
This will most likely be solved with the new resource syntax coming in an alpha soon :tm:

bradleyharden, 2021-06-02:
We discussed this on the Matrix channel. To `static`ally declare a value, you **must** specify the full, concrete type. Someone has to do that, and it can't really be RTIC, so it must fall to the user. Unless there is some way to use inferred types with `static` variables, this is not possible.

closed 2021-06-02 by bradleyharden