rtic-rs/rtic #391: NVIC: SCR and CCR unimplemented

jiayihu, 2020-10-12:
I'm getting started to using Rust and RTIC, so this might just be an error from me probably. However I had the following working code:

```rust
#![no_std]
#![no_main]

use cortex_m::peripheral::syst::SystClkSource;
use cortex_m_rt::entry;
use cortex_m_semihosting::{debug, hprint};
use hal::prelude::*;
use panic_semihosting as _;
use stm32f4::stm32f429::{interrupt, Interrupt, NVIC};
use stm32f4xx_hal as hal;

#[entry]
fn main() -> ! {
    hprint!("Hello world").unwrap();

    let cortex_peri = cortex_m::Peripherals::take().unwrap();
    let device_peri = hal::stm32::Peripherals::take().unwrap();

    let rcc = device_peri.RCC.constrain();
    setup_clocks(rcc);

    let mut syst = cortex_peri.SYST;
    let mut nvic = cortex_peri.NVIC;

    nvic.enable(Interrupt::EXTI0);

    syst.set_clock_source(SystClkSource::Core);
    syst.set_reload(180_000_000);
    syst.clear_current();
    syst.enable_counter();

    // exit_qemu();
    loop {
        while !syst.has_wrapped() {}

        NVIC::pend(Interrupt::EXTI0);
    }
}
```

But changing it to the following, which just tries to integrate RTIC, doesn't work anymore:

```rust
use cortex_m::peripheral::syst::SystClkSource;
use cortex_m_semihosting::{debug, hprint};
use hal::prelude::*;
use panic_semihosting as _;
use stm32f4::stm32f429::{interrupt, Interrupt, NVIC};
use stm32f4xx_hal as hal;

#[rtic::app(device = stm32f4xx_hal::stm32, peripherals = true)]
const APP: () = {
    #[init]
    fn init(cx: init::Context) {
        hprint!("Hello world").unwrap();

        let cortex_peri: cortex_m::Peripherals = cx.core;
        let device_peri: hal::stm32::Peripherals = cx.device;

        let rcc = device_peri.RCC.constrain();
        setup_clocks(rcc);

        let mut syst = cortex_peri.SYST;
        let mut nvic = cortex_peri.NVIC;

        nvic.enable(Interrupt::EXTI0);

        syst.set_clock_source(SystClkSource::Core);
        syst.set_reload(180_000_000);
        syst.clear_current();
        syst.enable_counter();

        // exit_qemu();
        loop {
            while !syst.has_wrapped() {}

            NVIC::pend(Interrupt::EXTI0);
        }
    }
};
```

It prints `(qemu) NVIC: SCR and CCR unimplemented` on QEMU, whereas on the device it hungs

TeXitoi, 2020-10-12:
https://rtic.rs/0.5/book/en/by-example/timer-queue.html

I suspect the **important** section is somethat related.

TeXitoi, 2020-10-12:
On the device, nothing will happen, as interrupts are masked in `init`

jiayihu, 2020-10-13:
I've been trying with the device but with no luck. Isn't unmasking sufficient for the following code?

```rust
#[rtic::app(device = stm32f4xx_hal::stm32, peripherals = true)]
const APP: () = {
    #[init]
    fn init(cx: init::Context) {
        hprintln!("Hello world").unwrap();

        unsafe {
            NVIC::unmask(Interrupt::EXTI0);
        }
        rtic::pend(Interrupt::EXTI0);
    }

    #[idle]
    fn idle(_: idle::Context) -> ! {
        hprintln!("Idle").unwrap();
        rtic::pend(Interrupt::EXTI0);

        loop {}
    }

    #[task(binds = EXTI0)]
    fn exti0(_: exti0::Context) {
        hprint!(".").unwrap();
    }
};
```

I see that it prints the two dots from software `rtic::pend` but manually pushing the button has no effect

korken89, 2020-10-15:
Hi, in RTIC you should not be touching the NVIC as RTIC sets it all up.
In your original example you should move the loop to the idle funtion.

If you want the button to generate an interrupt you need to set the button up using the HAL or the PAC. :)

jiayihu, 2020-10-15:
Thanks, clear enough for me 😄

closed 2020-10-15 by jiayihu