rtic-rs/rtic #417: `#[task_local]` resources should use `'static` lifetimes, not the priority lifetime

bradleyharden, 2020-11-16:
I ran into a problem this morning. I have a resource struct that looks something like this:

```rust
#[resources]
struct Resources {
    #[task_local]
    one: One,
    #[task_local]
    #[init(None)]
    two: Option<&'static mut One>,
}
```

And I want to use it like this

```rust
#[task(resources = [one, two])]
fn some_task(ctx: some_task::Context) {
    let some_task::Resources { one, two } = some_task.resources;
    *two = Some(one);
}
```

I don't see any reason why this shouldn't be possible, yet I wasn't able to do it with RTIC. The resulting code looks like this:

```rust
static mut __rtic_internal_one: core::mem::MaybeUninit<One> = core::mem::MaybeUninit::uninit();
static mut __rtic_internal_two: Option<&'static mut One> = None;

pub struct Resources<'a> {
    pub one: &'a mut One,
    pub two: &'a mut Option<&'static mut One>,
}

impl<'a> Resources<'a> {
    #[inline(always)]
    pub unsafe fn new(priority: &'a rtic::export::Priority) -> Self {
        Resources {
            one: &mut *__rtic_internal_one.as_mut_ptr(),
            two: &mut __rtic_internal_two,
        }
    }
}
```

The problem is that the lifetimes of fields `one` and `two` are both `'a`, which comes from the `priority`. But given that they are `#[task_local]`, there is no need to restrict the lifetime of the reference. Really, both fields should be `'static` references.

Indeed, if I modify the generated struct to use `'static` references and remove the lifetime parameter, my code works just fine.

```rust
pub struct Resources<'a> {
    pub one: &'static mut One,
    pub two: &'static mut Option<&'static mut One>,
}

impl Resources {
    #[inline(always)]
    pub unsafe fn new(priority: &rtic::export::Priority) -> Self {
        Resources {
            one: &mut *__rtic_internal_one.as_mut_ptr(),
            two: &mut __rtic_internal_two,
        }
    }
}
```

My guess is that it would be easy to change`#[task_local]` resources to use `'static` lifetimes. The bigger problem comes when a `Resources` struct uses *only* `#[task_local]` resources. Then, the lifetime parameter is superfluous, and the compiler complains. Removing it means changing the generated code for `some_task::Resources::new` and `some_task::Context`. Alternatively, you could add a `PhantomData` field for the unused lifetime, but that has the unintended side effect of preventing users from destructuring to retrieve resources.

bradleyharden, 2020-11-16:
Hmm, actually, I think that might not be right. If `two` were a shared resource, rather than a `#[task_local]` resource, then other tasks could use it to access the non-shared resource `one`. The only way to prevent that is to restrict the lifetime of `&mut One` so that it can't be put into `two` in the first place, which is exactly what the current code does.

Could someone else take a look at this and see if it makes sense at all? It may not.

This problem came up in a real use case, and I think I just realized a better way to solve it that circumvents all of these issues.

perlindgren, 2020-11-17:
Thanks for the comments, the use case you describe is quite important, and we need to find a sound solution to it. We will look into it asap.
/Per

perlindgren, 2020-11-17:
In RTIC you define for each task exactly what resources it can/may access. And there should be only a singe owning variable of each mutable location. In your example:

``` Rust
#[task(resources = [one, two])]
fn some_task(ctx: some_task::Context) {
    let some_task::Resources { one, two } = some_task.resources;
    *two = Some(one);
}
```

There will be two ways to reach `one` (after it has been run once). This looks like UB right.

There might be other cases though, requiring `&`static mut`. E.g. if you want to allow `init` to split a statically allocated queue into a sender/receiver pair. I don't think we can do that now. As `init` is a one time task, the above problem cannot occur, and I think it will be sound.
/Per

korken89, 2020-11-24:
Hi, we discussed this at today's meeting, and it can be UB to have `'static` lifetime in tasks. 
It would only be acceptable in `init` which is only run once.

If you have more questions about this, reopen this or make a new issue :)

closed 2020-11-24 by korken89