rtic-rs/rtic #433: Code Gen Panics When there is `monotonic` is defined but no task defined

Gentoli, 2020-12-06:
If an app is defined with `monotonic`
```
#[rtic::app(... , monotonic = rtic::cyccnt::CYCCNT,...)]
```
and no task is defined, the procedural macro fails with 
``` log
help: message: called `Option::unwrap()` on a `None` value
```
on this line:
https://github.com/rtic-rs/cortex-m-rtic/blob/e65a86f6962218b66a14b7ccc40eec7976880eba/macros/src/codegen/pre_init.rs#L79

This is a niche issue but could be encountered when creating new projects. There should be a proper error message for this.

korken89, 2020-12-06:
Hi, this will disappear with the require of monotonic and will be fixed in `0.6.0-alpha.1`.
Thanks for bringing it to our attention!

bradleyharden, 2020-12-29:
Just ran into this myself. Looking forward to the next release

PR #436 (2021-03-04): New monotonic
Design document: https://hackmd.io/vWa9GvssR8qBfUYgMZm0CQ

Closes #433 
Closes #432
Closes #427
Closes #426 
Closes #403
Closes #332
Closes #312 
Closes #309 
Closes #299 
Closes #292
Closes #247
Closes #219
closed 2021-03-04 by bors[bot]