rtic-rs/rtic #434: Specifying hardware priorities used by rtic

dthoma, 2020-12-09:
I'm using the nordic softdevice together with rtic. The softdevice reserves certain interrupt priorities, using those in the application can result in a crash. I managed to figure out how rtic maps logical priorities to the hardware, but that is not very userfriendly. It would be great to have an API to specify, which priorities can be used and how tasks are mapped to them.

The priorities reserved by the softdevice are not continous. The most significant priorities are reserved for time critical tasks in the BLE-stack are reserved, than there is a gap and then a lower priority is reserved as well. So it can be important whether rtic tasks are assigned to priorities below or above that lower priority.

perlindgren, 2020-12-09:
Priorities in RTIC are logic priorities (1 being the lowest priority task, 0 being the `idle`). Huge readability advantage over the bitchifted and inverted HW priorities.

If your target if your target has 16 priorities, and you know that the softdevice uses the highest 3 (15,14,13) and the lowest (1), make sure you use only priorities between 2-12. In RTIC you actually get 17 (0-16) priorities on a 16 priority HW (by disabling interrupts for the highest priority). The NVIC HW you gets you priorities 0..15 (if you don't do the disable interrupts trick of course...) 

With that said, you cannot really "reserve" priorities, many tasks/interrupts can have the same priority (its just a priority after all, on a tie the HW would dispatch the interrupt with the lowest vector index first). 

So I'm not really sure what you mean by "reserve". You can check the generated code (`cargo expand`, or in the `target` folder, to see exactly what RTIC does, to make sure it does not disable the interrupt sources, and make sure that the sources are properly enabled (not sure how the "soft device" does that...) And, of course you cannot use an interrupt vector tied to the same vector number for a task in RTIC. 

Ultimately one could write an RTIC friendly binding to "soft device" where the "soft device" interrupts are bound in RTIC (and just trampoline into the "soft device" code accordingly). Overhead is very low for such a trampoline, so it shouldn't be a problem.

So my wild guess here is that its not the priorities that is the problem, but rather the binding of interrupts. 

/Per

dthoma, 2020-12-09:
By "reserve" I mean the specification says, you must not use them. If you do, it causes timing issues, at least for the highest ones. They are used for tasks that must not be interrupted by other tasks form the application. So from my understanding, if rtic uses those priorities for some other interrupt, that might happen, causing the softdevice to detect a timing issue and stop.

Of course one can do what you suggest: read the code of rtic, figure out, how it assigns priorities, map the hardware priorities back to the logic priorities and avoid those. That is what I did and it works. This is a feature request, because it is cumbersome to do that.

Writing bindings does not really help. First of all, softdevice does not have a public api to do those callbacks anyway, so this would need to be reverse engineered and could change at any time. But then one still would need to avoid certain priorities and the application would crash randomly if by accident someone would not.

What I am thinking of is something like an attribute to declare priorities that may not be used. Would even be an improvement if this would be declared in terms of logical priorities. It would be enough to just fail at compile time if such a priority is used. I don't even see the need to map the logical priorities round "reserved" ones.

But it would be great to be able to specify such a constraint that has to be satisfied for the application to work. It is too easy to mess that up, especially, as it might not be noticeable immediately but cause random crashes. So I just wondered, what could be done to make that safer.

AfoHT, 2020-12-10:
Interesting!

Within RTIC there is an option how task priorities will be arranged, and the default is to "compress" them leaving no gaps. I think if we would allow disabling this the user would be fully responsible for assigning tasks to interrupts and RTIC would not "translate" that afterwards.

See [macros/src/lib.rs](https://github.com/rtic-rs/cortex-m-rtic/blob/e65a86f6962218b66a14b7ccc40eec7976880eba/macros/src/lib.rs#L23)

and the [implementation](https://github.com/rtic-rs/rtic-syntax/blob/4121a0206a9946d756d79c5b56f547209ab784b5/src/optimize.rs#L5)

Currently this setting is not user-accessible.

If I understand the issue correctly you would like to "blacklist" certain priorities, which then would cause an error alerting the user that this priority is not allowed.

Would providing users with a use-case example/walkthrough of how to solve this problem "manually" be sufficient? Your hardware and its special setup could showcase the issue and solution.

The function converting between logical and hardware/NVIC priorities found in [export.rs](https://github.com/rtic-rs/cortex-m-rtic/blob/e65a86f6962218b66a14b7ccc40eec7976880eba/src/export.rs#L181) for reference:
```
pub fn logical2hw(logical: u8, nvic_prio_bits: u8) -> u8 {
    ((1 << nvic_prio_bits) - logical) << (8 - nvic_prio_bits)
}
```

Not sure where to go with this in general, but I think allowing the user full control of the used priorities is a good start.

dthoma, 2020-12-10:
> Within RTIC there is an option how task priorities will be arranged, and the default is to "compress" them leaving no gaps.
Oh, I did not know about that part. This means my code might actually still cause timings issues and it just seems to work by coincidence. So I would probably have to declare rtic tasks with the priorities I do not want to use, to keep them free.

> If I understand the issue correctly you would like to "blacklist" certain priorities, which then would cause an error alerting the user that this priority is not allowed.
Exactly.

> Would providing users with a use-case example/walkthrough of how to solve this problem "manually" be sufficient? Your hardware and its special setup could showcase the issue and solution.
Yeah, for me personally some documentation of what rtic does (and possibly a way to disable the optimization of priorities) would be enough.

Just for completeness, this is the documentation of the so called softdevice I'm refering to:
https://infocenter.nordicsemi.com/topic/sds_s112/SDS/s1xx/processor_avail_interrupt_latency/exception_mgmt_sd.html

I assume there are many other usecase out there though, were some external piece of software has to run alongside an rtic application.

korken89, 2021-03-04:
This is resolved now

closed 2021-03-04 by korken89