rtic-rs/rtic #438: Trying to run schedule example on stm32f4, compiler has no idea what [SSI0] is

GregWoods, 2021-01-03:
I have seen some recent examples use:
```#[app(device = stm32f4xx_hal::stm32, monotonic = rtic::cyccnt::CYCCNT, dispatchers = [SSI0], peripherals = true)]```
which results in: ```error: expected literal```

Older code seems to use
```
    extern "C" {
        fn SSI0();
    }
```
at the end of main.rs, but that produces the error: 
```error[E0599]: no variant or associated item named `SSI0` found for enum `stm32f4xx_hal::stm32::Interrupt` in the current scope```

There seems to be very little written on how to use RTIC on anything but the lm3s6965, so any help would be appreciated.

perlindgren, 2021-01-03:
Hi Greg.

The dispatchers syntax is experimental for the next upcoming 0.6x. 

You can find examples of 0.5x for some hw platforms at:

https://github.com/rtic-rs/rtic-examples

Let us know on the RTIC matrix if you run into some problems.
/Per

GregWoods, 2021-01-04:
Thanks for the pointer back to rtic-examples. I'd forgotten about that repo, and I was only using the /examples folder in the rtic repo itself. Thanks to an stm32l4 example... the magic incantation I needed was:
```
    extern "C" {
        fn EXTI0();
    }
```

closed 2021-01-04 by GregWoods