rtic-rs/rtic #439: Multilock cosumes the mutex

bradleyharden, 2021-01-12:
The following code compiles, but adding the commented code gives an error. It looks like the mutex objects are moved and can no longer be accessed. This is not a problem using locks normally.

```rust
#![no_std]
#![no_main]

use panic_halt as _;
use atsamd_hal::target_device as pac;

#[rtic::app(device = crate::pac)]
mod app {

    #[resources]
    struct Resources {
        #[init(0)]
        one: u8,
        #[init(0)]
        two: u16,
    }

    #[init]
    fn init(_: init::Context) -> init::LateResources {
        init::LateResources{}
    }

    #[task(binds = PAC, priority = 1, resources = [one, two])]
    fn task_1(ctx: task_1::Context) {
        let task_1::Resources { one, two } = ctx.resources;
        (one, two).lock(|one, two| {
            *one = 5;
            *two = 10;
        });
        //(one, two).lock(|one, two| {
        //    *one = 5;
        //    *two = 10;
        //});
    }

    #[task(binds = AC, priority = 2, resources = [one, two])]
    fn task_2(ctx: task_2::Context) {
        let task_2::Resources { mut one, mut two } = ctx.resources;
        one.lock(|one| {
            two.lock(|two| {
                *one = 5;
                *two = 10;
            });
        });
        one.lock(|one| {
            two.lock(|two| {
                *one = 5;
                *two = 10;
            });
        });
    }

}
```

perlindgren, 2021-01-12:
I didn't test the multilock extensively. Seems to be like you say, the lock proxy is somehow consumed. We can bring it up on tonight's RTIC meeting. 

I see some other difference though. task_2 is on highest priority and you have deconstructed resource proxies as mut.

bradleyharden, 2021-01-12:
@perlindgren, `mut` resource proxies is unrelated. Originally, they were `mut` in both tasks, but the compiler complained that they didn't need to be `mut` in task_1. That's a separate issue I've also noticed. The resources need to be `mut` when using normal locks, but not when using multilock.

Don't the tasks need to be different priorities? Or are locks always required unless you explicitly mark a resource as `#[lock_free]`?

perlindgren, 2021-01-12:
In 0.6 we should provide symmetric API (and behavior) independent of priority. So there are some underlying bug lurking.

korken89, 2021-01-12:
@bradleyharden I have had a look, and looking at the type signature of the lock it accepts any type that implements lock.
And how you are showing the usage will move `one, two` into the tuple, which after the lock is complete, is not used anymore and goes out of scope.

I'm not sure on how to fix this from the tuple lock side. I need to experiment some.

korken89, 2021-01-13:
@bradleyharden We could require the tuple lock implementation to always take references, however this would incur an ergonomic cost for other users.
So my current recommendation is to simply use `&mut` references to resources,:

```rust
    #[task(binds = PAC, priority = 1, resources = [one, two])]
    fn task_1(ctx: task_1::Context) {
        let task_1::Resources { one, two } = ctx.resources;
        (&mut one, &mut two).lock(|one, two| {
            *one = 5;
            *two = 10;
        });
        (&mut one, &mut two).lock(|one, two| {
            *one = 5;
            *two = 10;
        });
    }
```

perlindgren, 2021-01-13:
I don't think that looks too bad. We could have both right. So if you pass without ref it will consume, with mutable reference not.

perlindgren, 2021-01-26:
We consider this closed, please re-open if the suggested solution is found insufficient.

closed 2021-01-26 by perlindgren